Universal-Debloater-Alliance/universal-android-debloater-next-generation #17: samsung SM-S9110: some bloatware are not listed

IceCodeNew, 2023-10-28:
**Describe the bug**
1. package `com.netflix.mediaclient` is preloaded on samsung SM-S9110
[image]

2. package `com.netflix.mediaclient` can be found from file `uad_lists.json` inside the cache directory of UAD
[image]

3. package `com.netflix.mediaclient` CAN NOT be found in the UAD even if all the filters are disabled
[image]

**Expected behavior**
package `com.netflix.mediaclient` be shown in the UAD

**You have a solution?**
I got no idea how this happened

**UAD log**

<details>
  <summary>Click me</summary>

```
2023-10-28 18:37:08 ERROR [src/core/config.rs:79] Failed to read config file: `No such file or directory (os error 2)`
2023-10-28 18:37:08 ERROR [src/core/config.rs:81] Restoring default config file
2023-10-28 18:37:08 ERROR [src/core/sync.rs:71] ADB: No such file or directory (os error 2)
2023-10-28 18:37:10 ERROR [src/core/sync.rs:71] ADB: No such file or directory (os error 2)
2023-10-28 18:37:10 ERROR [src/core/sync.rs:329] get_device_list() -> ADB was not found
2023-10-28 18:37:10 ERROR [src/core/sync.rs:71] ADB: No such file or directory (os error 2)
2023-10-28 18:37:10 ERROR [src/core/sync.rs:329] get_device_list() -> ADB was not found
2023-10-28 18:37:11 ERROR [src/core/sync.rs:71] ADB: No such file or directory (os error 2)
2023-10-28 18:37:11 ERROR [src/core/sync.rs:329] get_device_list() -> ADB was not found
2023-10-28 18:37:11 ERROR [src/core/sync.rs:71] ADB: No such file or directory (os error 2)
2023-10-28 18:37:11 ERROR [src/core/sync.rs:329] get_device_list() -> ADB was not found
2023-10-28 18:37:12 ERROR [src/core/sync.rs:71] ADB: No such file or directory (os error 2)
2023-10-28 18:37:12 ERROR [src/core/sync.rs:329] get_device_list() -> ADB was not found
2023-10-28 18:37:12 ERROR [src/core/sync.rs:71] ADB: No such file or directory (os error 2)
2023-10-28 18:37:12 ERROR [src/core/sync.rs:329] get_device_list() -> ADB was not found
2023-10-28 18:37:13 ERROR [src/core/sync.rs:71] ADB: No such file or directory (os error 2)
2023-10-28 18:37:13 ERROR [src/core/sync.rs:329] get_device_list() -> ADB was not found
2023-10-28 18:37:13 ERROR [src/core/sync.rs:71] ADB: No such file or directory (os error 2)
2023-10-28 18:37:13 ERROR [src/core/sync.rs:329] get_device_list() -> ADB was not found
2023-10-28 18:40:08 DEBUG [src/gui/views/settings.rs:77] Config change: Settings { general: GeneralSettings { theme: "Dark", expert_mode: false }, device: DeviceSettings { device_id: "", disable_mode: false, multi_user_mode: false, backup: BackupSettings { backups: [], selected: None, users: [], selected_user: None, backup_state: "" } } }
2023-10-28 18:40:08 DEBUG [src/core/config.rs:65] config: New device settings saved
2023-10-28 18:40:59 INFO  [src/gui/views/list.rs:125] -----------------------------------------------------------------
2023-10-28 18:40:59 INFO  [src/gui/views/list.rs:126] ANDROID_SDK: 0 | DEVICE: fetching devices...
2023-10-28 18:40:59 INFO  [src/gui/views/list.rs:130] -----------------------------------------------------------------
2023-10-28 18:41:00 ERROR [src/gui/views/list.rs:716] AppsView ready but no phone found
2023-10-28 18:41:30 INFO  [src/gui/views/list.rs:125] -----------------------------------------------------------------
2023-10-28 18:41:30 INFO  [src/gui/views/list.rs:126] ANDROID_SDK: 33 | DEVICE: samsung SM-S9110
2023-10-28 18:41:30 INFO  [src/gui/views/list.rs:130] -----------------------------------------------------------------
2023-10-28 20:02:57 INFO  [src/core/sync.rs:123] [Recommended] pm uninstall --user 0 com.android.hotwordenrollment.okgoogle -> Success
2023-10-28 20:02:57 INFO  [src/core/sync.rs:123] [Recommended] pm uninstall --user 0 com.android.hotwordenrollment.xgoogle -> Success
2023-10-28 20:02:57 INFO  [src/core/sync.rs:123] [Recommended] pm uninstall --user 0 com.android.chrome -> Success
2023-10-28 20:02:57 INFO  [src/core/sync.rs:123] [Recommended] pm uninstall --user 0 com.android.providers.partnerbookmarks -> Success
2023-10-28 20:02:57 INFO  [src/core/sync.rs:123] [Recommended] pm uninstall --user 0 com.android.traceur -> Success
2023-10-28 20:02:57 INFO  [src/core/sync.rs:123] [Recommended] pm uninstall --user 0 com.facebook.appmanager -> Success
2023-10-28 20:02:57 INFO  [src/core/sync.rs:123] [Recommended] pm uninstall --user 0 com.facebook.system -> Success
2023-10-28 20:02:57 INFO  [src/core/sync.rs:123] [Recommended] pm uninstall --user 0 com.facebook.services -> Success
2023-10-28 20:02:58 INFO  [src/core/sync.rs:123] [Recommended] pm uninstall --user 0 com.google.android.apps.tachyon -> Success
2023-10-28 20:02:58 INFO  [src/core/sync.rs:123] [Recommended] pm uninstall --user 0 com.google.android.googlequicksearchbox -> Success
2023-10-28 20:02:58 INFO  [src/core/sync.rs:123] [Recommended] pm uninstall --user 0 com.google.android.printservice.recommendation -> Success
2023-10-28 20:02:58 INFO  [src/core/sync.rs:123] [Recommended] pm uninstall --user 0 com.google.android.projection.gearhead -> Success
2023-10-28 20:02:58 INFO  [src/core/sync.rs:123] [Recommended] pm uninstall --user 0 com.microsoft.skydrive -> Success
2023-10-28 20:02:58 INFO  [src/core/sync.rs:123] [Recommended] pm uninstall --user 0 com.samsung.android.aircommandmanager -> Success
2023-10-28 20:02:58 INFO  [src/core/sync.rs:123] [Recommended] pm uninstall --user 0 com.samsung.android.allshare.service.mediashare -> Success
2023-10-28 20:02:58 INFO  [src/core/sync.rs:123] [Recommended] pm uninstall --user 0 com.samsung.android.app.watchmanagerstub -> Success
2023-10-28 20:02:58 INFO  [src/core/sync.rs:123] [Recommended] pm uninstall --user 0 com.samsung.android.ardrawing -> Success
2023-10-28 20:02:59 INFO  [src/core/sync.rs:123] [Recommended] pm uninstall --user 0 com.samsung.android.aremoji -> Success
2023-10-28 20:02:59 INFO  [src/core/sync.rs:123] [Recommended] pm uninstall --user 0 com.samsung.android.aremojieditor -> Success
2023-10-28 20:02:59 INFO  [src/core/sync.rs:123] [Recommended] pm uninstall --user 0 com.samsung.android.beaconmanager -> Success
2023-10-28 20:02:59 INFO  [src/core/sync.rs:123] [Recommended] pm uninstall --user 0 com.samsung.android.bixby.wakeup -> Success
2023-10-28 20:02:59 INFO  [src/core/sync.rs:123] [Recommended] pm uninstall --user 0 com.samsung.android.easysetup -> Success
2023-10-28 20:02:59 INFO  [src/core/sync.rs:123] [Recommended] pm uninstall --user 0 com.samsung.android.rubin.app -> Success
2023-10-28 20:02:59 INFO  [src/core/sync.rs:123] [Recommended] pm uninstall --user 0 com.samsung.android.sdk.handwriting -> Success
2023-10-28 20:02:59 INFO  [src/core/sync.rs:123] [Recommended] pm uninstall --user 0 com.samsung.android.svcagent -> Success
2023-10-28 20:02:59 INFO  [src/core/sync.rs:123] [Recommended] pm uninstall --user 0 com.sec.android.app.chromecustomizations -> Success
2023-10-28 20:03:00 INFO  [src/core/sync.rs:123] [Recommended] pm uninstall --user 0 com.swiftkey.swiftkeyconfigurator -> Success
2023-10-28 20:03:00 INFO  [src/core/sync.rs:123] [Recommended] pm uninstall --user 0 com.sec.android.mimage.avatarstickers -> Success
2023-10-28 20:09:19 INFO  [src/gui/mod.rs:248] -----------------------------------------------------------------
2023-10-28 20:09:19 INFO  [src/gui/mod.rs:249] ANDROID_SDK: 33 | DEVICE: samsung SM-S9110
2023-10-28 20:09:19 INFO  [src/gui/mod.rs:253] -----------------------------------------------------------------
2023-10-28 20:09:21 INFO  [src/gui/views/list.rs:125] -----------------------------------------------------------------
2023-10-28 20:09:21 INFO  [src/gui/views/list.rs:126] ANDROID_SDK: 33 | DEVICE: samsung SM-S9110
2023-10-28 20:09:21 INFO  [src/gui/views/list.rs:130] -----------------------------------------------------------------
2023-10-28 20:09:51 WARN  [src/core/uad_lists.rs:194] Could not load remote debloat list: https://raw.githubusercontent.com/0x192/universal-android-debloater/main/resources/assets/uad_lists.json: Network Error: timed out reading response
2023-10-28 20:09:55 ERROR [src/gui/views/list.rs:721] Error loading remote debloat list for the phone. Fallback to embedded (and outdated) list
```
</details>

IceCodeNew, 2023-10-28:
Version of UAD: 0.5.2 (I downloaded the artifacts from the [Build and release](https://github.com/Frigyes06/universal-android-debloater/actions/workflows/release.yml) pipeline)
https://github.com/Frigyes06/universal-android-debloater/actions/runs/6662236187

Frigyes06, 2023-10-28:
Interesting. I do see in the debug that it's trying to get the uad list from the inactive repo, so I'll need to fix that.

Frigyes06, 2023-10-28:
@IceCodeNew Could you check this build: https://github.com/Frigyes06/universal-android-debloater/actions/runs/6678792589
I think I fixed it in #18

Frigyes06, 2023-10-28:
Oh nevermind, I found the real culprit.

When listing packages, UAD uses `pm list packages -s`
-s filters output to system packages only

Will update PR soon

Frigyes06, 2023-10-28:
New build run, with fix: https://github.com/Frigyes06/universal-android-debloater/actions/runs/6678966992

closed 2023-10-28 by Frigyes06

Frigyes06, 2023-10-28:
Also, this was such a big oversight that I'm releasing a new version and bumping minor version number

PR #18 (2023-10-28): ADB get all packages fix
Fixes #17