Universal-Debloater-Alliance/universal-android-debloater-next-generation #28: [Suggestion] Github Organization

Alliegaytor, 2023-11-01:
The original project being left unmaintained is what caused this fork to be created, what is to say that it won't happen again? A github organization with a few maintainers on it would *hopefully* prevent that from happening. I don't know who would be willing to maintain this or how people should be chosen, but maybe something could be added in the README about this?

Frigyes06, 2023-11-01:
I think I answered this question on the issue below this one. If I see contributors who know what they're doing and are responsible, I'm more than happy to make them a maintainer.

Frigyes06, 2023-11-01:
Org created. Gonna push a new release with new urls soon

AnonymousWP, 2023-11-02:
I volunteer for being an active maintainer of the repo. Submitted a lot of PRs and when I'm back from holidays I'll be able to actively contribute again.

Frigyes06, 2023-11-02:
> I volunteer for being an active maintainer of the repo. Submitted a lot of PRs and when I'm back from holidays I'll be able to actively contribute again.

I sent an invite 👍

closed 2023-11-03 by Alliegaytor