Universal-Debloater-Alliance/universal-android-debloater-next-generation #31: [bug] Weird window border in Wayland

yozachar, 2023-11-02:
**Describe the bug**

Window border does not follow system design.

[image]

**Expected behavior**

Render system themed title-bar.

[image]

**UAD log**

<details><summary>Log</summary>

```console
$ RUST_BACKTRACE=full uad_gui-opengl
2023-04-25 15:06:24 WARN  [src/gui/mod.rs:102] Trying to download remote UAD list
2023-04-25 15:06:25 WARN  [src/gui/views/list.rs:142] No custom selection imported: No such file or directory (os error 2)
warning: queue 0x7efc58000ca0 destroyed while proxies still attached:
  zwp_primary_selection_offer_v1@4278190081 still attached
  wl_data_offer@4278190080 still attached
  zwp_primary_selection_device_v1@47 still attached
  zwp_primary_selection_device_manager_v1@40 still attached
  wl_data_device@46 still attached
  wl_seat@45 still attached
  wl_data_device_manager@44 still attached
  wl_registry@43 still attached
warning: queue 0x55eadbcb0c20 destroyed while proxies still attached:
  wl_buffer@60 still attached
  wl_buffer@61 still attached
  wl_buffer@62 still attached
  wl_shm_pool@63 still attached
  wl_buffer@38 still attached
  wl_buffer@37 still attached
  wl_buffer@36 still attached
  wl_buffer@35 still attached
  wl_buffer@34 still attached
  wl_subsurface@29 still attached
  wl_surface@28 still attached
  wl_subsurface@27 still attached
  wl_surface@26 still attached
  wl_subsurface@25 still attached
  wl_surface@24 still attached
  wl_subsurface@23 still attached
  wl_surface@22 still attached
  wl_subsurface@21 still attached
  wl_surface@20 still attached
  xdg_wm_base@18 still attached
  wl_surface@17 still attached
  xdg_activation_v1@12 still attached
  zwp_text_input_manager_v3@11 still attached
  zwp_pointer_constraints_v1@10 still attached
  zwp_relative_pointer_manager_v1@9 still attached
  wl_seat@8 still attached
  wl_subcompositor@7 still attached
  wl_output@6 still attached
  wl_shm@5 still attached
  wl_compositor@4 still attached
  wl_registry@2 still attached
thread 'main' panicked at 'called `Result::unwrap_err()` on an `Ok` value: ()', src/gui/mod.rs:452:29
stack backtrace:
   0:     0x55eada651b9e - <unknown>
   1:     0x55eada4cdffc - <unknown>
   2:     0x55eada6252d4 - <unknown>
   3:     0x55eada652ab1 - <unknown>
   4:     0x55eada6539d0 - <unknown>
   5:     0x55eada653592 - <unknown>
   6:     0x55eada653506 - <unknown>
   7:     0x55eada6534f0 - <unknown>
   8:     0x55eada4cb9f2 - <unknown>
   9:     0x55eada4d29c2 - <unknown>
  10:     0x55eada6cf3cf - <unknown>
  11:     0x55eada6f8bca - <unknown>
  12:     0x55eada6ce605 - <unknown>
  13:     0x7efcd5677790 - <unknown>
  14:     0x7efcd567784a - __libc_start_main
  15:     0x55eada49ba6e - <unknown>
  16:                0x0 - <unknown>
```

</details>

Frigyes06, 2023-11-02:
Definitely try the latest version. If the problem persists, please let me know

yozachar, 2023-11-02:
I tried both `uad_gui-noseflupdate-linux` & `uad_gui-noseflupdate-linux-opengl` pre-release. The retro-styled-window is gone, but it the newer one does not follow system theme:

[image]

Also, window screenshot does not capture the border / title bar, weird.

[image]

---

Logs. I do not have ADB installed yet, you can ignore that error.

```
2023-11-02 10:08:06 ERROR [src/core/sync.rs:71] ADB: No such file or directory (os error 2)
2023-11-02 10:08:06 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.14.1/src/gles/egl.rs:808] Re-initializing Gles context due to Wayland window
2023-11-02 10:08:06 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.14.1/src/gles/egl.rs:808] Re-initializing Gles context due to Wayland window
2023-11-02 10:08:06 ERROR [src/core/sync.rs:71] ADB: No such file or directory (os error 2)
2023-11-02 10:08:06 ERROR [src/core/sync.rs:329] get_device_list() -> ADB was not found
2023-11-02 10:08:07 ERROR [src/core/sync.rs:71] ADB: No such file or directory (os error 2)
2023-11-02 10:08:07 ERROR [src/core/sync.rs:329] get_device_list() -> ADB was not found
warning: queue 0x7f4c0c000ca0 destroyed while proxies still attached:
  zwp_primary_selection_offer_v1@4278190081 still attached
  wl_data_offer@4278190080 still attached
  zwp_primary_selection_device_v1@54 still attached
  zwp_primary_selection_device_manager_v1@47 still attached
  wl_data_device@53 still attached
  wl_seat@51 still attached
  wl_data_device_manager@50 still attached
  wl_registry@45 still attached
warning: queue 0x56063fb7edc0 destroyed while proxies still attached:
  wl_buffer@64 still attached
  wl_buffer@63 still attached
  wl_buffer@70 still attached
  wl_shm_pool@69 still attached
  wl_subsurface@29 still attached
  wl_surface@28 still attached
  wl_subsurface@27 still attached
  wl_surface@26 still attached
  wl_subsurface@25 still attached
  wl_surface@24 still attached
  wl_subsurface@23 still attached
  wl_surface@22 still attached
  wl_subsurface@21 still attached
  wl_surface@20 still attached
  xdg_wm_base@18 still attached
  xdg_activation_v1@12 still attached
  zwp_text_input_manager_v3@11 still attached
  zwp_pointer_constraints_v1@10 still attached
  zwp_relative_pointer_manager_v1@9 still attached
  wl_seat@8 still attached
  wl_subcompositor@7 still attached
  wl_output@6 still attached
  wl_shm@5 still attached
  wl_compositor@4 still attached
  wl_registry@2 still attached
```

Frigyes06, 2023-11-02:
Wayland user :)
I wonder if it works right on Xorg

yozachar, 2023-11-02:
[image]

Yup, on Xorg, it picks up GTK3/2 title bar and theme, and none of those warnings (apart from ADB) is seen in the logs.

I do not have Nvidia.

Frigyes06, 2023-11-02:
Super weird. Wayland is quirky like that. I think we should keep the issue open, but I personally won't be dealing with it for now. You could maybe try building the PR with iced deps bumped...

Frigyes06, 2023-11-03:
Can you also try the build at: https://github.com/Universal-Debloater-Alliance/universal-android-debloater/actions/runs/6748710018

yozachar, 2023-11-06:
> Can you also try the build at: https://github.com/Universal-Debloater-Alliance/universal-android-debloater/actions/runs/6748710018

Same windows border:

[image]

When `adb` is installed, *no messages* are flushed to the console. *Otherwise* you can still see:

```console
warning: queue 0x7fadbc000ca0 destroyed while proxies still attached:
  zwp_primary_selection_offer_v1@4278190083 still attached
  wl_data_offer@4278190082 still attached
  zwp_primary_selection_device_v1@46 still attached
  zwp_primary_selection_device_manager_v1@39 still attached
  wl_data_device@45 still attached
  wl_seat@44 still attached
  wl_data_device_manager@43 still attached
  wl_registry@42 still attached
warning: queue 0x55ca132eae30 destroyed while proxies still attached:
  wl_buffer@49 still attached
  wl_buffer@51 still attached
  wl_shm_pool@50 still attached
  wl_subsurface@29 still attached
  wl_surface@28 still attached
  wl_subsurface@27 still attached
  wl_surface@26 still attached
  wl_subsurface@25 still attached
  wl_surface@24 still attached
  wl_subsurface@23 still attached
  wl_surface@22 still attached
  wl_subsurface@21 still attached
  wl_surface@20 still attached
  xdg_wm_base@18 still attached
  xdg_activation_v1@12 still attached
  zwp_text_input_manager_v3@11 still attached
  zwp_pointer_constraints_v1@10 still attached
  zwp_relative_pointer_manager_v1@9 still attached
  wl_seat@8 still attached
  wl_subcompositor@7 still attached
  wl_output@6 still attached
  wl_shm@5 still attached
  wl_compositor@4 still attached
  wl_registry@2 still attached
```

Frigyes06, 2023-11-06:
I get the same warning on my wayland, but my border works fine. The queue error is an issue in driver/software the gui uses, not our codebase.

lavafroth, 2024-01-08:
```sh
WAYLAND_DISPLAY=wayland-1 cargo r
```

yozachar, 2024-01-13:
```console
$ WAYLAND_DISPLAY=wayland-1 uad
```

This gives GTK3 window:

[image]

Frigyes06, 2024-01-13:
> ```
> $ WAYLAND_DISPLAY=wayland-1 uad
> ```

That fixes the queue errors for me. Also the app is is so much smaller on wayland. Weird

AnonymousWP, 2024-03-17:
Is this fix easy to implement? Any updates on this?

Frigyes06, 2024-03-17:
> Is this fix easy to implement? Any updates on this?

We don't know what's going on, so no. It's not easy to implement

yozachar, 2024-04-05:
[image]

Looks like this has been fixed!

```
Gnome 46
UAD v1.0.3
```

AnonymousWP, 2024-04-05:
Closing this issue as it has been fixed (likely in an update of Wayland?)

closed 2024-04-05 by AnonymousWP