Universal-Debloater-Alliance/universal-android-debloater-next-generation #33: Crash when starting

LupSpie, 2023-11-03:
**Describe the bug**
Attempting to start UAD results in a crash.

**Expected behavior**
For the program to start normally

**UAD log**
```
$ RUST_BACKTRACE=full ./uad_gui-linux-opengl
2023-11-03 01:38:58 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.14.1/src/gles/egl.rs:808] Re-initializing Gles context due to Wayland window
2023-11-03 01:38:58 ERROR [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-0.14.2/src/backend/direct.rs:894] Error in Adapter::request_device: Limit 'max_compute_workgroups_per_dimension' value 65535 is better than allowed 0
2023-11-03 01:38:58 ERROR [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-0.14.2/src/backend/direct.rs:894] Error in Adapter::request_device: Limit 'max_compute_workgroups_per_dimension' value 65535 is better than allowed 0
warning: queue 0x555fa092bdc0 destroyed while proxies still attached:
xdg_wm_base@20 still attached
zwp_text_input_manager_v3@13 still attached
wl_output@12 still attached
xdg_activation_v1@11 still attached
wl_subcompositor@10 still attached
zwp_relative_pointer_manager_v1@9 still attached
zwp_pointer_constraints_v1@8 still attached
wl_seat@7 still attached
wl_shm@6 still attached
zxdg_decoration_manager_v1@5 still attached
wl_compositor@4 still attached
wl_registry@2 still attached
Error: GraphicsCreationFailed(GraphicsAdapterNotFound)
```

Frigyes06, 2023-11-03:
What version? Did you try the non-opengl version?

LupSpie, 2023-11-03:
0.6.0
I tried both versions and the error was the same

Frigyes06, 2023-11-03:
Can you try Xorg instead of Wayland?

LupSpie, 2023-11-03:
No difference. Same error, but without the wayland stuff on the log

```log
2023-11-03 14:23:47 ERROR [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-0.14.2/src/backend/direct.rs:894] Error in Adapter::request_device: Limit 'max_compute_workgroups_per_dimension' value 65535 is better than allowed 0
2023-11-03 14:23:47 ERROR [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-0.14.2/src/backend/direct.rs:894] Error in Adapter::request_device: Limit 'max_compute_workgroups_per_dimension' value 65535 is better than allowed 0
```

LupSpie, 2023-11-03:
Surprisingly that did fix it :partying_face:

Frigyes06, 2023-11-03:
Yeah, appears I have debloated the dependencies too much. Gonna push a new release soon.

PR #35 (2023-11-03): OpenGL Fix
Fixes #33
closed 2023-11-03 by Frigyes06