Universal-Debloater-Alliance/universal-android-debloater-next-generation #36: UAD doesn't open

Dv5n, 2023-11-04:
**Describe the bug**
The UAD GUI doesn't open and I see in Task Manager the ADB.EXE at 0% CPU Usage with the command line: "adb -L tcp:5037 fork-server server --reply-fd 504"

**Expected behavior**
UAD should open and display the GUI.

**You have a solution?**
No.

**UAD log**
.

Dv5n, 2023-11-04:
> Maybe try [uad_gui-windows-opengl.exe](https://github.com/Universal-Debloater-Alliance/universal-android-debloater/releases/download/0.6.0/uad_gui-windows-opengl.exe)?

I tried it.

Frigyes06, 2023-11-04:
Any logs you could post?

Dv5n, 2023-11-04:
> Any logs you could post?

How to see the log?

ProtonOSS, 2023-11-04:
> > Any logs you could post?
> 
> How to see the log?

Log location: `%localappdata%\uad`
[image]

Dv5n, 2023-11-06:
> > > Any logs you could post?
> > 
> > 
> > How to see the log?
> 
> Setting location `%appdata%\uad` Log location: `%localappdata%\uad` [image]

Thanks, here's the log

"2023-11-06 14:51:36 WARN  [C:\Users\runneradmin\.cargo\registry\src\index.crates.io-6f17d22bba15001f\wgpu-hal-0.14.1\src\vulkan\adapter.rs:1069] Missing extensions: ["VK_KHR_maintenance2", "VK_KHR_storage_buffer_storage_class", "VK_EXT_sampler_filter_minmax", "VK_KHR_timeline_semaphore"]"

Dv5n, 2023-11-06:
> I'm not an expert but it looks like vulkan drivers are missing?
> By my understanding the `VK` stands for `Vulkan` so it's related, but not sure about the error or how to solve it.

I think it's because i have some problems with OpenGL/Vulkan.

I was having some problems in the past, SM64EX-Coop with OpenGL didn't open. But after an update to Windows, it fixed my problems.

But i think there's still a problem, Minecraft (after my pc gets a little bit hot) it glitches the graphics.

And when it happens it scares me.

Dv5n, 2023-11-06:
But it's weird because i have set W: as the main windows directory.

But i still have the C: for extra compatibility.

Dv5n, 2023-11-06:
Thanks, Now works perfectly.
I can finally destroy Netflix from my samsung galaxy A20e.

closed 2023-11-06 by Frigyes06