Universal-Debloater-Alliance/universal-android-debloater-next-generation #37: Can't unistall com.motorola.ccc.devicemanagement

felisucoibi, 2023-11-06:
**Your phone model**:
motorola g73 5g
**Describe the issue**
I click to unisntall it and it does not work.

The apps com.motorola.ccc.devicemanagement and the app com.motorola.paks can't be deleted or disable. Should warn somehow that can't do the job and why.

AnonymousWP, 2023-11-23:
Why didn't you use [an issue template](https://github.com/Universal-Debloater-Alliance/universal-android-debloater/issues/new/choose)? It's pretty unclear what you exactly want us to do now. Also, what did you try to fix the issue?

kyoforkshomebrews, 2023-12-12:
It's probably the same story as with com.huawei.parentcontrol, can't be disabled/removed even via ADB cause the system strictly blocks it.

### com.motorola.ccc.devicemanagement
**List**: `Oem`
**Removal**: `Recommended`
> Mobile Device Management (MDM) allows company’s IT department to reach inside your phone in the background, allowing them to ensure
your device is secure, know where it is, and remotely erase your data if the phone is stolen.
You should NEVER install a MDM tool on your phone. Never.\nhttps://onezero.medium.com/dont-put-your-work-email-on-your-personal-phone-ef7fef956c2f\nhttps://blog.cdemi.io/never-accept-an-mdm-policy-on-your-personal-phone/
>
>Uninstalling or disabling might not work on some devices.

In com.motorola.paks' description it says "ADB: Package Protected." I'm guessing that refers to this issue anyway so it's not needed to change.

I think @felisucoibi is also suggesting to have a modal pop up when such an error happens with any app.

felisucoibi, 2023-12-13:
> I think @felisucoibi is also suggesting to have a modal pop up when such an error happens with any app.

Exactly, now seems the app is broken and not working no error no nothing just dont remove it.

Frigyes06, 2023-12-17:
I'll see what I can do. I'm still learning the UI library and the current code.

Frigyes06, 2024-02-28:
This will need to be a new modal in the gui, which is a lot of work. Unsure when I can get around to implementing it. Maybe @lavafroth or @adhirajsinghchauhan can take a look?

adhirajsinghchauhan, 2024-03-01:
Right now there's many "error" states that aren't even logged at all, so first we'd need to finish up that work, and only then would it make sense to show errors in the UI.

Personally, I'm not interested in doing logging/error-related stuff, but I can help out with anything UI.

lavafroth, 2024-03-24:
Hi, is there a definitive way to reproduce an error so that I can test my local branch?

Frigyes06, 2024-05-03:
@AnonymousWP I think this is fixed by #409

AnonymousWP, 2024-05-13:
> @AnonymousWP I think this is fixed by #409

Yeah I think you're right. Closing this issue.

closed 2024-05-13 by AnonymousWP