Universal-Debloater-Alliance/universal-android-debloater-next-generation #41: DELETE_FAILED_DEVICE_POLICY_MANAGER

Drjacky, 2023-11-10:
`ERROR [src/core/utils.rs:228] [Unlisted] pm uninstall --user 0 com.example -> Failure [DELETE_FAILED_DEVICE_POLICY_MANAGER]`

Frigyes06, 2023-11-11:
What package are you trying to uninstall?

Drjacky, 2023-11-11:
> What package are you trying to uninstall?

A system app (which might have a protected package/folder)

Frigyes06, 2023-11-22:
This is not a bug in the program. Best we can do is add that it cannot be uninstalled into the package's description

Frigyes06, 2023-11-23:
Can you tell us which app you were trying to uninstall, so we can add info to it's description?

Drjacky, 2023-11-24:
> Can you tell us which app you were trying to uninstall, so we can add info to it's description?

It's a custom app from a custom AOSP.

Frigyes06, 2023-11-24:
Is it on the list?
I need to know what you want us to fix to close this issue

Drjacky, 2023-11-25:
https://stackoverflow.com/a/54021398/421467 -> https://stackoverflow.com/a/5387620/421467

So, it's an admin app (I think you can simulate this issue by installing [Avast](https://play.google.com/store/apps/details?id=com.avast.android.mobilesecurity)) but the issue with my admin app is the "disabling the admin checkbox" is disabled; I cannot disable that admin app.

AnonymousWP, 2023-11-27:
Try disabling instead of uninstalling. As you may know, you can't uninstall system packages; you can only disable them. Try that instead.

Drjacky, 2023-11-27:
> Try disabling instead of uninstalling. As you may know, you can't uninstall system packages; you can only disable them via ADB. Try that instead.

"Disable Switchbox" is disabled.

AnonymousWP, 2023-11-27:
> > Try disabling instead of uninstalling. As you may know, you can't uninstall system packages; you can only disable them via ADB. Try that instead.
> 
> "Disable Switchbox" is disabled.

Huh? What do you mean with "switchbox"? You're supposed to tick the box "disable instead of uninstall" (or something like that) in the settings.

Drjacky, 2023-11-30:
> Huh? What do you mean with "switchbox"? You're supposed to tick the box "disable instead of uninstall" (or something like that) in the settings.

I mean in the OS, the factory of the Android Device has disabled the ability to disable the Admin app.

AnonymousWP, 2023-11-30:
> I mean in the OS, the factory of the Android Device has disabled the ability to disable the Admin app.

Don't use that one on your phone, but disable the app via UAD.

Drjacky, 2023-12-02:
> Don't use that one on your phone, but disable the app via UAD.

This is what brought me to create this issue. It shows this error:
```bash
ERROR [src/core/utils.rs:228] [Unlisted] pm uninstall --user 0 com.example -> Failure [DELETE_FAILED_DEVICE_POLICY_MANAGER]
```

AnonymousWP, 2023-12-03:
Is your phone a managed phone by the company you work at? In that case, due to Android policies you can't disable/uninstall certain apps.

closed 2023-12-03 by Drjacky