Universal-Debloater-Alliance/universal-android-debloater-next-generation #43: Cannot execute on Linux

poach3r, 2023-11-10:
**Describe the bug**
Attempting to execute the program using a bash shell fails with the following error:
bash: ./uad_gui-linux: cannot execute: required file not found

Attempting to execute the program using a zsh shell fails with the following error:
zsh: no such file or directory: ./uad_gui-linux

**Expected behavior**
UAD executes

Note: I have given the file execute permissions

Frigyes06, 2023-11-11:
Are you in the right directory?

poach3r, 2023-11-12:
Yes.
poacher@nixos ~ cd - ~/Downloads/
poacher@nixos ~/Downloads - ./uad_gui-linux
zsh: no such file or directory: ./uad_gui-linux

poach3r, 2023-11-16:
I already gave it execute permissions, I wrote this in the original issue
```
drwxr-xr-x  2 poacher users 4.0K Nov 16 16:51 .
drwx------ 25 poacher users 4.0K Nov 16 16:51 ..
-rw-r--r--  1 poacher users 2.4G Nov  9 21:08 nixos-gnome-23.05.4738.41de143fda10-x86_64-linux.iso
-rwxr-xr-x  1 poacher users 9.8M Nov 12 13:43 uad_gui-linux
```

Skallwar, 2023-11-17:
Are you running NixOS ?

poach3r, 2023-11-28:
Yes

Skallwar, 2023-11-29:
Then I guess it's not about UAD itself but more about the fact that NixOS stores the dynamic linker in a non standard path.  See https://nixos.wiki/wiki/Packaging/Binaries

Frigyes06, 2024-01-03:
@poach3r Did this ever work out for you?

markifi, 2024-01-17:
i have the same issue. 

`
[user@nixos:~]$ nix-env -iA nixos.android-tools
warning: name collision in input Nix expressions, skipping '/home/user/.nix-defexpr/channels_root/nixos'
suggestion: remove 'nixos' from either the root channels or the user channels
installing 'android-tools-34.0.4'
building '/nix/store/67jc3s8y5yasifi7f1i04gqmdzipz314-user-environment.drv'...

[...]

[user@nixos:~/Downloads/uad]$ chmod +x uad_gui-linux 

[user@nixos:~/Downloads/uad]$ ./uad_gui-linux 
bash: ./uad_gui-linux: cannot execute: required file not found

[user@nixos:~/Downloads/uad]$ ls -lah
total 9,4M
drwxr-xr-x 2 user users 4,0K jan   17 17.36 .
drwxr-xr-x 8 user users 116K jan   17 17.36 ..
-rwxr-xr-x 1 user users 9,3M 2022 júl    3 uad_gui-linux

[user@nixos:~/Downloads/uad]$ hostnamectl
     Static hostname: nixos
           Icon name: computer-laptop
             Chassis: laptop 💻
          Machine ID: b5dbfd0038954efdb040f240b1187598
             Boot ID: d9b1673ac3aa41d6ad91ab2d5ccfbb76
    Operating System: NixOS 23.11 (Tapir)             
      OS Support End: Sun 2024-06-30
OS Support Remaining: 5month 1w 5d
              Kernel: Linux 6.1.71
        Architecture: x86-64
`

Frigyes06, 2024-01-17:
@markifi Magyarok?

Also you'll have to compile from source, see above. NixOS weirdness

Skallwar, 2024-01-18:
A quick and dirty way that probably work for the binary is to use `steam-run uad_gui-linux`

ilovethensa, 2024-01-19:
> A quick and dirty way that probably work for the binary is to use `steam-run uad_gui-linux`

Couldnt do it 
```bash
[nix-shell:~/Downloads]$ steam-run uad_gui-linux
/nix/store/i93vxva1m34ch2pq1790i4zzdfifl0q0-steam-run: line 21: exec: uad_gui-linux: not found
```

ilovethensa, 2024-01-19:
Ill try to write a flake.nix for this project

lavafroth, 2024-02-12:
Hey everyone, please check out the `nix-flake` branch. You can run `nix develop` in the workdir to compile uad-ng.

LudovicoPiero, 2024-02-14:
The `universal-android-debloater` is already available on nixpkgs. You can use it; however, it's still on version 0.6.2, though
```console
nix-shell -p universal-android-debloater
uad_gui
```

Frigyes06, 2024-02-25:
> The `universal-android-debloater` is already available on nixpkgs. You can use it; however, it's still on version 0.6.2, though
> ```console
> nix-shell -p universal-android-debloater
> uad_gui
> ```

Maybe we can ping the maintainer to get up to date?

Frigyes06, 2024-02-25:
I pinged the nix maintainer, they'll bump version soon. Closing this issue now.

closed 2024-02-25 by Frigyes06

afifurrohman-id, 2025-02-25:
now the binary name is:
```sh
uad-ng
```