Universal-Debloater-Alliance/universal-android-debloater-next-generation #49: Open links and copy

Skizzy24, 2023-11-19:
Make that you can open links from the app description and copy the app package or app description.

adhirajsinghchauhan, 2023-12-28:
iced has no support for this, and it won't come anytime soon: https://github.com/iced-rs/iced/issues/36.

There's two ways to copy text if using iced:
1. TextInput but with overridden styles (it's a text input field, so we'll need to to make it look like normal text)
2. Manage selection state ourselves

I don't like (1) because its just ugly to do. A GUI toolkit doesn't support (normal) text selection? Yuck. And then there'd be accessibility concerns, with screen readers for example, but I have no idea if iced even handles that at all.

(2) would increase the amount of code we'd need to maintain, with little actual benefit.

Frigyes06, 2024-01-03:
Yeah, this is really a missing feature in iced rather than UAD-ng. It'll have to be a wontfix until we make a new ui or iced implements copiable text

MathewJake365, 2024-06-12:
This feature is very useful. And without this we have to type package names and long links by looking at them. Please make this happen.

Frigyes06, 2024-07-25:
See: https://github.com/iced-rs/iced/issues/36

mtxadmin, 2024-12-19:
It's really weird that such an essential function as copying text is ignored by Iced's devs for years.

Did you think to rewrite this GUI on a different framework, instead of Iced?

adhirajsinghchauhan, 2024-12-20:
Iced has more problems than just not supporting text-copy. For example, it still does not have [any concept of accessibility](https://github.com/iced-rs/iced/issues/552). It may not be important to you or me, but it is important to someone out there.

Anyway, while there may be another framework that supports everything we think we want, it will certainly be lacking in other areas where iced shines. It's less about a framework problem, and more just that Rust, as a language, was never really meant for GUIs. It's great for systems programming, and other parallels where memory safety is essential, but outside of that? It's nothing special.

Over time Rust GUIs may have parity with Qt (C++/Python), but even then, they simply cannot compare to the vast featureset of web-based GUIs . Devs who work on web UIs never have to worry about compromising in that respect. Sure, some folks would argue that Electron (web) and familiars add a lot of bloat, and aren't the best for performance, but most people don't care about the difference between 10MB and 100MB size, or 20MB RAM and 350MB, or 1% CPU usage and 5%.

(side note: Compose is looking real great over on the Kotlin/Java side of things, but once again there is a compromise on binary size)

However, in this project it is essential that size & performance is prioritized. After all, it's a debloater tool. Would be odd if its binary is itself bloated. So, we don't really have a choice. We either make the monumental effort to switch over to Qt, or stick with Rust + iced.

I think it's time we bite the bullet and just implement any of the hacky workarounds [I mentioned](https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/issues/49#issuecomment-1870779538) to get this issue resolved. We can't wait for frameworks to do what we want; we'll have to do it ourselves (and maybe contribute upstream). As for which dev is motivated to do that, well, we'll have to see.

dcampbell24, 2025-06-21:
Well you can make the link a button and use [webbrowser](https://github.com/amodm/webbrowser-rs) to open links.

samuele-ruffino96, 2025-07-04:
> iced has no support for this, and it won't come anytime soon: [iced-rs/iced#36](https://github.com/iced-rs/iced/issues/36).
> 
> There's two ways to copy text if using iced:
> 
>     1. TextInput but with overridden styles (it's a text input field, so we'll need to to make it look like normal text)
> 
>     2. Manage selection state ourselves
> 
> 
> I don't like (1) because its just ugly to do. A GUI toolkit doesn't support (normal) text selection? Yuck. And then there'd be accessibility concerns, with screen readers for example, but I have no idea if iced even handles that at all.
> 
> (2) would increase the amount of code we'd need to maintain, with little actual benefit.

In [that example](https://github.com/iced-rs/iced/issues/36#issuecomment-3035400175), I showed a way to obtain text selection capabilities. Not sure if it could be useful for your use case.

Rudxain, 2025-07-04:
> Well you can make the link a button and use [webbrowser](https://github.com/amodm/webbrowser-rs) to open links.

We already have
https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/blob/725bfe183cee97505421709dfdc824d90bd4c0e0/src/core/utils.rs#L125-L142
but thanks for mentioning an alt!

dcampbell24, 2025-07-04:
It is a bit more robust than just hard coding one option to use as an opener.

samuele-ruffino96, 2025-07-09:
Jus opened a [PR](https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/pull/1010).  Hope it can be a useful starting point.

closed 2025-07-10 by Rudxain