Universal-Debloater-Alliance/universal-android-debloater-next-generation #50: The lists contains user apps

Skizzy24, 2023-11-19:
**Describe the bug**
I was uninstalling apps in the recommended list when it showed the Google Docs app, but its not a system app so i can uninstall it myself

**Expected behavior**
The program does show only system apps

**You have a solution?**
No

Frigyes06, 2023-11-19:
This is not a bug, but a feature. The program is supposed to list user apps too. We could add a toggle to turn it on and off maybe?

Frigyes06, 2023-12-11:
> I feel like it used to only show system apps? I'd also find that more logical... don't need a Bloatware remover to remove those anyway..

It did. Newest release (pending) reverts it to system apps only.

AnonymousWP, 2023-12-12:
> > I feel like it used to only show system apps? I'd also find that more logical... don't need a Bloatware remover to remove those anyway..
> 
> It did. Newest release (pending) reverts it to system apps only.

Can you link your PR to this issue?

commit b03260a6 (2024-04-19): [Feature] Persistent settings (#44 / fix #50)
closed 2023-12-13 by AnonymousWP