Universal-Debloater-Alliance/universal-android-debloater-next-generation #51: Ok google search not working

Redhawk18, 2023-11-20:
**Samsung s8**:

**Describe the issue**
I actually use this feature and would love to have it again.

**You have a solution?**
I've enabled both 
```
com.android.hotwordenrollment.okgoogle
com.android.hotwordenrollment.xgoogle
```
But that seems to just be voice detection, not the actual service.

**UAD log**
```rs
2023-11-20 14:04:02 DEBUG [src/core/update.rs:178] Checking for UAD update
2023-11-20 14:04:02 WARN  [src/gui/mod.rs:102] Trying to download remote UAD list
2023-11-20 14:04:23 ERROR [src/core/utils.rs:228] [ADB] reboot -> adb: no devices/emulators found
2023-11-20 14:04:27 DEBUG [src/core/update.rs:178] Checking for UAD update
2023-11-20 14:04:27 WARN  [src/gui/mod.rs:102] Trying to download remote UAD list
2023-11-20 14:05:28 INFO  [src/gui/mod.rs:457] -----------------------------------------------------------------
2023-11-20 14:05:28 INFO  [src/gui/mod.rs:458] ANDROID_SDK: 0 | PHONE: fetching devices...
2023-11-20 14:05:28 WARN  [src/gui/views/list.rs:142] No custom selection imported: No such file or directory (os error 2)
2023-11-20 14:35:55 DEBUG [src/core/update.rs:178] Checking for UAD update
2023-11-20 14:35:55 WARN  [src/gui/mod.rs:102] Trying to download remote UAD list
2023-11-20 14:35:55 INFO  [src/gui/mod.rs:457] -----------------------------------------------------------------
2023-11-20 14:35:55 INFO  [src/gui/mod.rs:458] ANDROID_SDK: 28 | PHONE: samsung SM-G950U
2023-11-20 14:35:55 WARN  [src/gui/views/list.rs:142] No custom selection imported: No such file or directory (os error 2)
2023-11-20 14:47:39 INFO  [src/gui/mod.rs:457] -----------------------------------------------------------------
2023-11-20 14:47:39 INFO  [src/gui/mod.rs:458] ANDROID_SDK: 28 | PHONE: samsung SM-G950U
2023-11-20 14:47:40 WARN  [src/gui/views/list.rs:142] No custom selection imported: No such file or directory (os error 2)
2023-11-20 14:48:17 INFO  [src/core/utils.rs:222] [Recommended] pm uninstall --user 0 com.android.hotwordenrollment.okgoogle -> Success
2023-11-20 14:48:17 INFO  [src/core/utils.rs:222] [Recommended] pm uninstall --user 0 com.android.egg -> Success
2023-11-20 14:48:19 INFO  [src/core/utils.rs:222] [Recommended] pm uninstall --user 0 com.amazon.mShop.android.shopping -> Success
2023-11-20 14:48:19 INFO  [src/core/utils.rs:222] [Recommended] pm uninstall --user 0 com.android.bluetoothmidiservice -> Success
2023-11-20 14:48:20 INFO  [src/core/utils.rs:222] [Recommended] pm uninstall --user 0 com.amazon.appmanager -> Success
2023-11-20 14:48:20 INFO  [src/core/utils.rs:222] [Recommended] pm uninstall --user 0 com.android.dreams.phototable -> Success
2023-11-20 14:48:21 INFO  [src/core/utils.rs:222] [Recommended] pm uninstall --user 0 android.autoinstalls.config.samsung -> Success
2023-11-20 14:48:25 INFO  [src/core/utils.rs:222] [Recommended] pm uninstall --user 0 com.android.calllogbackup -> Success
2023-11-20 14:48:25 INFO  [src/core/utils.rs:222] [Recommended] pm uninstall --user 0 com.android.providers.partnerbookmarks -> Success
2023-11-20 14:49:05 INFO  [src/core/utils.rs:222] [Recommended] pm uninstall --user 0 com.android.chrome -> Success
2023-11-20 14:49:06 INFO  [src/core/utils.rs:222] [Recommended] pm uninstall --user 0 com.android.traceur -> Success
2023-11-20 14:49:06 INFO  [src/core/utils.rs:222] [Recommended] pm uninstall --user 0 com.android.hotwordenrollment.xgoogle -> Success
2023-11-20 14:49:07 INFO  [src/core/utils.rs:222] [Recommended] pm uninstall --user 0 com.asurion.android.verizon.vms -> Success
2023-11-20 14:50:22 INFO  [src/core/utils.rs:222] [Recommended] pm uninstall --user 0 com.cequint.ecid -> Success
2023-11-20 14:50:36 INFO  [src/core/utils.rs:222] [Recommended] pm uninstall --user 0 com.cnn.mobile.android.phone.edgepanel -> Success
2023-11-20 14:50:37 INFO  [src/core/utils.rs:222] [Recommended] pm uninstall --user 0 com.diotek.sec.lookup.dictionary -> Success
2023-11-20 14:50:38 INFO  [src/core/utils.rs:222] [Recommended] pm uninstall --user 0 com.dsi.ant.plugins.antplus -> Success
2023-11-20 14:50:38 INFO  [src/core/utils.rs:222] [Recommended] pm uninstall --user 0 com.dsi.ant.sample.acquirechannels -> Success
2023-11-20 14:50:39 INFO  [src/core/utils.rs:222] [Recommended] pm uninstall --user 0 com.enhance.gameservice -> Success
2023-11-20 14:50:55 INFO  [src/core/utils.rs:222] [Recommended] pm uninstall --user 0 com.google.android.apps.maps -> Success
2023-11-20 14:50:56 INFO  [src/core/utils.rs:222] [Recommended] pm uninstall --user 0 com.google.android.apps.restore -> Success
2023-11-20 14:50:57 INFO  [src/core/utils.rs:222] [Recommended] pm uninstall --user 0 com.google.android.feedback -> Success
2023-11-20 14:51:11 INFO  [src/core/utils.rs:222] [Recommended] pm uninstall --user 0 com.google.android.gm -> Success
2023-11-20 14:51:21 INFO  [src/core/utils.rs:222] [Recommended] pm uninstall --user 0 com.google.android.googlequicksearchbox -> Success
2023-11-20 14:51:23 INFO  [src/core/utils.rs:222] [Recommended] pm uninstall --user 0 com.google.android.marvin.talkback -> Success
2023-11-20 14:51:24 INFO  [src/core/utils.rs:222] [Recommended] pm uninstall --user 0 com.google.android.onetimeinitializer -> Success
2023-11-20 14:51:25 INFO  [src/core/utils.rs:222] [Recommended] pm uninstall --user 0 com.google.android.partnersetup -> Success
2023-11-20 14:51:26 INFO  [src/core/utils.rs:222] [Recommended] pm uninstall --user 0 com.google.android.printservice.recommendation -> Success
2023-11-20 14:51:42 INFO  [src/core/utils.rs:222] [Recommended] pm uninstall --user 0 com.google.android.youtube -> Success
2023-11-20 14:51:43 INFO  [src/core/utils.rs:222] [Recommended] pm uninstall --user 0 com.google.ar.core -> Success
2023-11-20 14:51:44 INFO  [src/core/utils.rs:222] [Recommended] pm uninstall --user 0 com.hancom.office.editor.hidden -> Success
2023-11-20 14:51:46 INFO  [src/core/utils.rs:222] [Recommended] pm uninstall --user 0 com.hiya.star -> Success
2023-11-20 14:51:47 INFO  [src/core/utils.rs:222] [Recommended] pm uninstall --user 0 com.knox.vpn.proxyhandler -> Success
2023-11-20 14:51:49 INFO  [src/core/utils.rs:222] [Recommended] pm uninstall --user 0 com.LogiaGroup.LogiaDeck -> Success
2023-11-20 14:51:50 INFO  [src/core/utils.rs:222] [Recommended] pm uninstall --user 0 com.monotype.android.font.chococooky -> Success
2023-11-20 14:51:51 INFO  [src/core/utils.rs:222] [Recommended] pm uninstall --user 0 com.monotype.android.font.cooljazz -> Success
2023-11-20 14:51:52 INFO  [src/core/utils.rs:222] [Recommended] pm uninstall --user 0 com.monotype.android.font.foundation -> Success
2023-11-20 14:51:52 INFO  [src/core/utils.rs:222] [Recommended] pm uninstall --user 0 com.monotype.android.font.rosemary -> Success
2023-11-20 14:51:53 INFO  [src/core/utils.rs:222] [Recommended] pm uninstall --user 0 com.qualcomm.atfwd -> Success
[... 178 lines cut ...]
2023-11-20 15:36:07 DEBUG [src/core/update.rs:178] Checking for UAD update
2023-11-20 15:36:07 WARN  [src/gui/mod.rs:102] Trying to download remote UAD list
2023-11-20 15:36:08 DEBUG [src/core/update.rs:178] Checking for UAD update
2023-11-20 15:36:08 WARN  [src/gui/mod.rs:102] Trying to download remote UAD list
2023-11-20 15:36:08 INFO  [src/gui/mod.rs:457] -----------------------------------------------------------------
2023-11-20 15:36:08 INFO  [src/gui/mod.rs:458] ANDROID_SDK: 28 | PHONE: samsung SM-G950U
2023-11-20 15:36:08 INFO  [src/gui/mod.rs:457] -----------------------------------------------------------------
2023-11-20 15:36:08 INFO  [src/gui/mod.rs:458] ANDROID_SDK: 28 | PHONE: samsung SM-G950U
2023-11-20 15:36:08 WARN  [src/gui/views/list.rs:142] No custom selection imported: No such file or directory (os error 2)
2023-11-20 15:36:08 WARN  [src/gui/views/list.rs:142] No custom selection imported: No such file or directory (os error 2)
2023-11-20 15:36:34 DEBUG [src/core/update.rs:178] Checking for UAD update
2023-11-20 15:38:08 DEBUG [src/core/update.rs:178] Checking for UAD update
2023-11-20 15:38:09 DEBUG [src/core/update.rs:178] Checking for UAD update
2023-11-20 15:38:14 DEBUG [src/core/update.rs:178] Checking for UAD update
2023-11-20 15:38:28 DEBUG [src/core/update.rs:178] Checking for UAD update
2023-11-20 15:39:05 DEBUG [src/core/update.rs:178] Checking for UAD update
2023-11-20 15:39:16 DEBUG [src/core/update.rs:178] Checking for UAD update
2023-11-20 15:39:16 WARN  [src/gui/mod.rs:102] Trying to download remote UAD list
[... cut ...]
2023-11-20 15:39:17 WARN  [src/gui/views/list.rs:142] No custom selection imported: No such file or directory (os error 2)
2023-11-20 15:39:33 INFO  [src/core/utils.rs:222] [Recommended] cmd package install-existing --user 0 com.android.hotwordenrollment.xgoogle -> Package com.android.hotwordenrollment.xgoogle installed for user: 0
2023-11-20 15:39:33 INFO  [src/core/utils.rs:222] [Recommended] cmd package install-existing --user 0 com.android.hotwordenrollment.okgoogle -> Package com.android.hotwordenrollment.okgoogle installed for user: 0

```

AnonymousWP, 2023-11-23:
Try enabling the following packages as well and see if that fixes it:

- `com.google.android.googlequicksearchbox`
- `com.google.android.apps.googleassistant`

Redhawk18, 2023-11-24:
> Try enabling the following packages as well and see if that fixes it:
> 
>     * `com.google.android.googlequicksearchbox`
> 
>     * `com.google.android.apps.googleassistant`

I actually don't have the 2nd one in my list, but restoring the first one fixed it. I hope someone can document apps like this.

closed 2023-11-24 by Redhawk18

Frigyes06, 2023-11-24:
#TODO: add description

PR #58 (2023-11-26): pkg(`com.google.android.googlequicksearchbox`): improve description
Fixes #51.