Universal-Debloater-Alliance/universal-android-debloater-next-generation #53: Error: GraphicsCreationFailed(GraphicsAdapterNotFound)

thomaswillecomme, 2023-11-23:
Hi ! thanks for the application fork an portage to arch !

I just got a phone completly bloated and since there is no lineageos support this is great ! i was struggling with an home made script before to get rid of all those fancy software.

Here is my problem. the application crash during initialization.

I'm running on Arch and installed from AUR using yay.
vulkan intel is installed
```
thomas@tuxbook(14:29:40) ~/documents/development $ pacman -Ss vulkan-intel
extra/vulkan-intel 1:23.2.1-2 [installé]
    Intel's Vulkan mesa driver
multilib/lib32-vulkan-intel 1:23.2.1-2 [installé]
    Intel's Vulkan mesa driver (32-bit)
```
i am using the intel chipset
`00:02.0 VGA compatible controller: Intel Corporation 2nd Generation Core Processor Family Integrated Graphics Controller (rev 09)`
and got an amd device which is broken and disabled
`01:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Whistler [Radeon HD 6730M/6770M/7690M XT]`

```
2023-11-23 14:23:51 ERROR [/home/thomas/.cache/yay/universal-android-debloater/src/cargo-home/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.15.4/src/vulkan/instance.rs:675] enumerate_adapters: Initialization of an object has failed
2023-11-23 14:23:51 WARN  [/home/thomas/.cache/yay/universal-android-debloater/src/cargo-home/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.15.4/src/gles/egl.rs:812] Re-initializing Gles context due to Wayland window
2023-11-23 14:23:51 ERROR [/home/thomas/.cache/yay/universal-android-debloater/src/cargo-home/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.15.4/src/vulkan/instance.rs:675] enumerate_adapters: Initialization of an object has failed
2023-11-23 14:23:51 ERROR [/home/thomas/.cache/yay/universal-android-debloater/src/cargo-home/registry/src/index.crates.io-6f17d22bba15001f/wgpu-0.15.1/src/backend/direct.rs:602] Error in Adapter::request_device: Limit 'max_compute_workgroups_per_dimension' value 65535 is better than allowed 0
2023-11-23 14:23:51 ERROR [/home/thomas/.cache/yay/universal-android-debloater/src/cargo-home/registry/src/index.crates.io-6f17d22bba15001f/wgpu-0.15.1/src/backend/direct.rs:602] Error in Adapter::request_device: Limit 'max_compute_workgroups_per_dimension' value 65535 is better than allowed 0
warning: queue 0x560272ff6a30 destroyed while proxies still attached:
  wl_buffer@37 still attached
  wl_buffer@36 still attached
  wl_buffer@35 still attached
  wl_buffer@34 still attached
  wl_subsurface@29 still attached
  wl_surface@28 still attached
  wl_subsurface@27 still attached
  wl_surface@26 still attached
  wl_subsurface@25 still attached
  wl_surface@24 still attached
  wl_subsurface@23 still attached
  wl_surface@22 still attached
  wl_subsurface@21 still attached
  wl_surface@20 still attached
  xdg_wm_base@18 still attached
  xdg_activation_v1@12 still attached
  zwp_text_input_manager_v3@11 still attached
  zwp_pointer_constraints_v1@10 still attached
  zwp_relative_pointer_manager_v1@9 still attached
  wl_seat@8 still attached
  wl_subcompositor@7 still attached
  wl_output@6 still attached
  wl_shm@5 still attached
  wl_compositor@4 still attached
  wl_registry@2 still attached
Error: GraphicsCreationFailed(GraphicsAdapterNotFound)

```

Frigyes06, 2023-11-23:
What package did you grab from the AUR? We're currently not maintaining an AUR package. It is most likely the old version. 
Also, are you using wayland or Xorg, and do you have an nVidia card or not?

Frigyes06, 2023-11-23:
Oh huh, someone actually did it!
https://aur.archlinux.org/packages/universal-android-debloater-opengl
https://aur.archlinux.org/packages/universal-android-debloater

I'll try to get in contact with the person maintaining it

Frigyes06, 2023-11-23:
It appears you're using Wayland, which is known to have issues with our software. #31 relates a bit too

Frigyes06, 2023-11-25:
Could you try Xorg?

thomaswillecomme, 2023-11-25:
Sorry, not right now. I broke my laptop and waiting for spare parts. Stuck with the phone. I made a 50 lines scripts that do the job by the way. I'll try to debug once m'y laptop is fixes

Frigyes06, 2024-02-28:
Any updates on this?

thomaswillecomme, 2024-03-10:
no sorry. Laptop not fixed and eventually i decided to go back to my old phone with lineage os for now.

AnonymousWP, 2024-03-10:
Closing due to the above.

closed 2024-03-10 by AnonymousWP