Universal-Debloater-Alliance/universal-android-debloater-next-generation #56: not able to uninstall any app

tmalai76, 2023-11-25:
**Describe the bug**
I'm having an issue with Redmi 3s Prime mobile.
It shows not installed for user 0. but in UAD it has only user 0, there are no other users shown.
Moreover, all the apps are shown only under unlisted.

**UAD log**
2023-11-25 13:35:46 INFO  [src\gui\views\list.rs:126] ANDROID_SDK: 23 | DEVICE: Xiaomi Redmi 3S
2023-11-25 13:35:46 INFO  [src\gui\views\list.rs:130] -----------------------------------------------------------------
2023-11-25 13:36:04 ERROR [src\core\sync.rs:120] [Unlisted] pm uninstall --user 0 com.xiaomi.payment
 -> Failure - not installed for 0
2023-11-25 13:36:05 ERROR [src\core\sync.rs:120] [Unlisted] pm uninstall --user 0 com.xiaomi.payment
 -> Failure - not installed for 0
2023-11-25 13:36:06 ERROR [src\core\sync.rs:120] [Unlisted] pm uninstall --user 0 com.xiaomi.payment
 -> Failure - not installed for 0
2023-11-25 13:36:06 ERROR [src\core\sync.rs:120] [Unlisted] pm uninstall --user 0 com.xiaomi.payment
 -> Failure - not installed for 0
2023-11-25 13:36:06 ERROR [src\core\sync.rs:120] [Unlisted] pm uninstall --user 0 com.xiaomi.payment
 -> Failure - not installed for 0
2023-11-25 13:36:06 ERROR [src\core\sync.rs:120] [Unlisted] pm uninstall --user 0 com.xiaomi.payment
 -> Failure - not installed for 0
2023-11-25 13:36:07 ERROR [src\core\sync.rs:120] [Unlisted] pm uninstall --user 0 com.xiaomi.payment
 -> Failure - not installed for 0

Frigyes06, 2024-01-03:
From the logs I'm seeing only one app trying to be uninstalled over and over. The error message is quite weird though. Is it still an issue in the latest release?

Frigyes06, 2024-02-28:
Please try the latest release and let us know if this is still an issue

AnonymousWP, 2024-03-03:
@tmalai76 This issue is going to be closed within a week if it's going to be stale.

tmalai76, 2024-03-04:
I'm sorry for not getting back to you sooner.
I have tried the latest version but the issue is still there. I tried to uninstall com.xiaomi.payment twice, but no response in the software. Please look at the following log file.

2024-03-04 23:39:07 WARN  [C:\Users\runneradmin\.cargo\registry\src\index.crates.io-6f17d22bba15001f\wgpu-hal-0.16.2\src\gles\egl.rs:464] 	EGL surface: -srgb
2024-03-04 23:39:07 WARN  [C:\Users\runneradmin\.cargo\registry\src\index.crates.io-6f17d22bba15001f\wgpu-hal-0.16.2\src\gles\egl.rs:226] No config found!
2024-03-04 23:39:07 WARN  [C:\Users\runneradmin\.cargo\registry\src\index.crates.io-6f17d22bba15001f\wgpu-hal-0.16.2\src\gles\egl.rs:215] EGL says it can present to the window but not natively
2024-03-04 23:39:07 DEBUG [src\core\update.rs:174] Checking for UAD-ng update
2024-03-04 23:39:12 INFO  [src\gui\views\list.rs:136] -----------------------------------------------------------------
2024-03-04 23:39:12 INFO  [src\gui\views\list.rs:137] ANDROID_SDK: 23 | DEVICE: Xiaomi Redmi 3S
2024-03-04 23:39:12 INFO  [src\gui\views\list.rs:141] -----------------------------------------------------------------
2024-03-04 23:39:43 ERROR [src\core\sync.rs:122] [Unlisted] pm uninstall --user 0 com.xiaomi.payment
 -> Failure - not installed for 0
2024-03-04 23:39:54 ERROR [src\core\sync.rs:122] [Unlisted] pm uninstall --user 0 com.xiaomi.payment
 -> Failure - not installed for 0

AnonymousWP, 2024-03-11:
@Frigyes06 Please follow up on this. I can't say for sure what's causing it.

AnonymousWP, 2024-03-18:
@tmalai76 Is there another user you can select in UAD-ng? Does that work?

Frigyes06, 2024-04-12:
Closing this as stale.

closed 2024-04-12 by Frigyes06