Universal-Debloater-Alliance/universal-android-debloater-next-generation #57: Messages and Calling breaks when disabling com.samsung.android.providers.contacts

Kharto, 2023-11-25:
Samsung s10e g970u

I disabled com.samsung.android.providers.contacts and my messaging and calling services would open but quickly and repeatedly crash. Upon enabling the same app they were usable again.

I don't know how to add this to the specific app package so I'm posting it here.

Frigyes06, 2023-11-26:
Was this on the list?
If not, could you send me all the unlisted packages so we can add them?

Kharto, 2023-11-26:
That was categorized under 'Expert'. The note reads: Likely same as com.android.providers.contacts, but for Samsung phones. Probably breaks contact functionality if disabled.

I'm very new to this. What's the best way to compile and send you the list of unlisted packages?

AnonymousWP, 2023-11-27:
I don't understand what you want us to do now actually. It's listed as expert, and the description clearly said it breaks contacts functionality. Do you want me to change the description? I opened a PR, please check/review it.

Kharto, 2023-11-27:
Yeah, it looks good. I just wanted to have it updated so that people would know that it breaks messaging and calling. Thanks!

closed 2023-11-27 by Kharto