Universal-Debloater-Alliance/universal-android-debloater-next-generation #67: Add disconnect functionality to send adb kill-server command for safe removal of devices

Ishatix, 2023-12-05:
See:
https://github.com/0x192/universal-android-debloater/issues/772

AnonymousWP, 2023-12-05:
What's the actual risk/issue here of disconnecting your phone by USB directly instead of first issuing `adb kill-server`?

Ishatix, 2023-12-06:
I think any risk would be dependent upon whatever other interactions might be going on in the background, e.g. say if you had an automatic sync service that runs when the phone is connected. A user may not always be aware or remember to keep such things mind. However, once you are able to "safely remove" your device, you can be confident that there is no ongoing activity that could be at risk. As matters stand, UAD prevents users from being able to safely remove their device without either rebooting it, or having adb running in parallel to manually send the kill-server command.

AnonymousWP, 2023-12-08:
> I think any risk would be dependent upon whatever other interactions might be going on in the background, e.g. say if you had an automatic sync service that runs when the phone is connected. A user [...]

`adb kill-server` doesn't 'safely' cancel any transfer processes either, as it's purely ADB-related, unless you're talking about performing an `adb push` command instead of using ADB? Wonder how much more safe it is than just unplugging the device _while_ you're transferring data through ADB.

AnonymousWP, 2024-03-01:
In my opinion, not really. I honestly also never use `adb kill-server` for disconnecting purposes, only for troubleshooting. Is it harmful to just disconnect your phone by USB when not using UAD-ng/ADB anymore?

Pombaroom, 2025-02-19:
Hi. Not sure if it's the same problem, but also somewhat about safe removal
I close UAD-NG, then I try to safe remove my phone by clicking the icon in the system tray, it says that I can't do that.
It is because adb.exe is still in processes even after I closed UAD-NG.
So it is strange for me - when I launch UAD-NG - it launchs adb.exe also, but when I close UAD-NG - it doesn't close adb.exe. Is it intentional? And why? Why do I need adb.exe proccess after I closed UAD-NG?

AnonymousWP, 2025-03-19:
I noticed the same. Although, it only consumes 1.3MB of RAM on my machine lol.