Universal-Debloater-Alliance/universal-android-debloater-next-generation #85: Rest of Huawei missing packages

kyoforkshomebrews, 2023-12-15:
**Your phone model:** Huawei P Smart 2019 (POT_LX1)

**Packages:**

```
com.huawei.android.overlay.modules.modulemetadata
com.huawei.android.dsdscardmanager
com.huawei.dsdscardmanager
com.huawei.hisuite
com.android.systemui.overlay

...
```

- [ ] **I removed all those packages on my phone**
      If not why. Leave the brackets blank and explain why.

I think apart from hisuite none of these can be safely removed without losing important features.

## Document each package the best you can

**List**: `Google`|`Misc`|`OEM` (manufacturer)|`AOSP`|`Pending`|`Carrier` (isp).

**Removal**: `Recommended`, `Advanced`, `Expert` (this can break important features),
or `Unsafe` (this can bootloop the phone or break extremely important features).

### com.android.systemui.overlay
**List**: `Oem`
**Removal**: `Unsafe`
> System UI Overlay. DO NOT remove this.
### com.huawei.hisuite
**List**: `Oem`
**Removal**: `Recommended`
> Can be uninstalled via settings. App to connect with the desktop program to Backup/Restore, flash firmware, etc. https://consumer.huawei.com/en/support/hisuite/
### com.huawei.android.overlay.modules.modulemetadata
**List**: `Oem`
**Removal**: `Unsafe`
> Module that contains metadata about the list of modules on the device. And that's about it. I wouldn't advise you to mess with it as it could break important modules (see W1nst0n/universal-android-debloater#37 on gitlab) Good explanation of what android modules are: https://www.xda-developers.com/android-project-mainline-modules-explanation/
### com.huawei.android.dsdscardmanager
**List**: `Oem`
**Removal**: `Expert`
> SIM management
### com.huawei.dsdscardmanager
**List**: `Oem`
**Removal**: `Expert`
> SIM management

AnonymousWP, 2023-12-16:
> **Your phone model:** Huawei P Smart 2019 (POT_LX1)
>
> **Packages:**
>
> ```
> com.huawei.android.overlay.modules.modulemetadata
> com.huawei.android.dsdscardmanager
> com.huawei.dsdscardmanager [...]

Is it by mistake that you didn't check the "I removed all those packages on my phone"?

kyoforkshomebrews, 2023-12-17:
To quote myself:
> I think apart from hisuite none of these can be safely removed without losing important features.

AnonymousWP, 2023-12-17:
> To quote myself:
> 
> > I think apart from hisuite none of these can be safely removed without losing important features.

Yes, I read that. Are you going to remove these packages or are you considering it? That way, we can be sure about its `removal` value.

closed 2024-02-25 by AnonymousWP