Universal-Debloater-Alliance/universal-android-debloater-next-generation #87: Bug after debloat

Damians0, 2023-12-16:
**Galaxy s23 ultra**:

**Describe the issue**
After having debloated my galaxy s23 ultra by choosing to delete all recommended applications and services, I am unable to take screenshots by pressing "capture screenshot" from the quick settings (drop-down menu).
 Furthermore, Google's voice dictation no longer works (a function that is invoked by pressing the microphone drawing on the keyboard).
 Which services do I need to reactivate to get these services working again? U.A.D 0.6.1

**You have a solution?**
What to do to fix the issue.

**UAD log**
Upload the logfile generated by UAD. It would be difficult to help you without it.

[image]
[image]

Frigyes06, 2023-12-16:
You should review each package's description before removing it in the future. You could search UAD list for "google dictation" and similar to find what package needs to be reenabled

AnonymousWP, 2024-01-03:
Closing due to inactivity.

closed 2024-01-03 by AnonymousWP