Universal-Debloater-Alliance/universal-android-debloater-next-generation #91: One UI 5/6 packages

Skizzy24, 2023-12-17:
**Packages:**

```
com.samsung.inputshare
com.samsung.android.scpm
com.samsung.android.widget.pictureframe
com.samsung.attribution
com.adaptivebrightnessgo
```

- [x] **I removed all those packages on my phone**

## Document each package the best you can

**List**: `Google`|`Misc`|`OEM` (manufacturer)|`AOSP`|`Pending`|`Carrier` (isp).

**Removal**: `Recommended`, `Advanced`, `Expert` (this can break important features),
or `Unsafe` (this can bootloop the phone or break extremely important features).

### \<com.samsung.inputshare\>

**List**: \<OEM\>
**Removal**: \<Recommended\>

> MultiControl,its use is to connect 2 devices and control one with the other.

### \<com.samsung.android.scpm\>

**List**: \<OEM\>
**Removal**: \<Recommended\>

> Samsung Cloud Platform Manager, related to samsung cloud, safe to remove if you dont use it.

### \<com.samsung.android.widget.pictureframe\>

**List**: \<OEM\>
**Removal**: \<Recommended\>

> Samsung gallery widget, shows your selected photos as widget.

### \<com.samsung.attribution\>

**List**: \<OEM\>
**Removal**: \<Recommended\>

> Related to data collection https://developer.samsung.com/galaxy-store/galaxy-store-statistics/user-attribution.html (?)

.### \<com.adaptivebrightnessgo\>

**List**: \<OEM\>
**Removal**: \<Advanced\>

> Not related to adaptive brightness in the sliding bar, the apk is called CameraLight, so i guess its for the camera.

AnonymousWP, 2023-12-19:
> com.samsung.inputshare
> Removal: <Recommended>
> MultiControl,its use is to connect 2 devices and control one with the other.

Why recommended if it can be a quite useful feature for some users actually? I changed it to Advanced for now (see linked PR).

Skizzy24, 2023-12-19:
> > com.samsung.inputshare
> > Removal: 
> > MultiControl,its use is to connect 2 devices and control one with the other.
> 
> Why recommended if it can be a quite useful feature for some users actually? I changed it to Advanced for now (see linked PR).

https://apps.samsung.com/appquery/appDetail.as?appId=com.samsung.android.inputshare&cId=000006157428
you can actually redownload it from the Galaxy Store and the desc says its for galaxy book an galaxy tab so ig pretty useless if you use a normal phone

AnonymousWP, 2023-12-19:
Yeah okay, fair point. Fixed (commit is still being pushed to the branch but is in the queue).

closed 2023-12-20 by Blitz517