Universal-Debloater-Alliance/universal-android-debloater-next-generation #98: OCRDataProvider

Skizzy24, 2023-12-18:
**Packages documentation to update:**

```
com.samsung.android.sdk.ocr
...
```

## Documentation Change

**List**: `Google`|`Misc`|`OEM` (manufacturer)|`AOSP`|`Pending`|`Carrier` (isp).
**Removal**: `Recommended`, `Advanced`, `Expert` (this can break important features),
or `Unsafe` (this can bootloop the phone or break extremely important features).

### \<OCRDataProvider\>

**Removal**: \<Recommended\>
:arrow_right: \<Advanced\>

### Proposed description

> It's for OCR, used among samsung apps like the gallery, without this, you can't extract text from images.

(the old description is very wrong)

AnonymousWP, 2023-12-19:
What does OCR stand for?

Skizzy24, 2023-12-19:
> What does OCR stand for?

OCR stands for "Optical Character Recognition." It is a technology that recognizes text within a digital image.

closed 2023-12-20 by Blitz517