Universal-Debloater-Alliance/universal-android-debloater-next-generation #99: Samsung ApexService

Skizzy24, 2023-12-18:
```
com.sec.android.app.apex
...
```

## Documentation Change

**List**: `Google`|`Misc`|`OEM` (manufacturer)|`AOSP`|`Pending`|`Carrier` (isp).
**Removal**: `Recommended`, `Advanced`, `Expert` (this can break important features),
or `Unsafe` (this can bootloop the phone or break extremely important features).

### \<com.sec.android.app.apex\>

**Removal**: \<Expert\> :arrow_right: \<Recommended\>

### Proposed description

> Without this motion photos will not work see https://www.samsung.com/au/support/mobile-devices/motion-photos , it is also for automatic useless story creation in the gallery app, these stories are random photos from your gallery and music.

AnonymousWP, 2023-12-19:
In my opinion this should be Advanced, as it breaks a feature some users may want to use. Changed. Let me know what you think.

Skizzy24, 2023-12-19:
> In my opinion this should be Advanced, as it breaks a feature some users may want to use. Changed. Let me know what you think.

yeah you are right

closed 2023-12-20 by Blitz517