Universal-Debloater-Alliance/universal-android-debloater-next-generation #112: some inbuilt are not getting uninstalled when uninstall button is clicked

vijayvardan, 2023-12-23:
Describe the bug:-
I have RealMe X2 mobile. Successfully deleted few apps like Youtube and Finshell.

Tick Checked the Box for deleting apps and clicked the uninstall button, but no action was noticed for these following apps:
1. com.heytap.market - App Market
2.  com.heytap.themestore - Theme Store
3. com.nearme.gamecenter - Game Center

//need a solution

AnonymousWP, 2023-12-25:
Post some logs and we can take a look at it.

Xevalous, 2024-01-04:
> Post some logs and we can take a look at it.

[image]

AnonymousWP, 2024-01-04:
> ![image](https://private-user-images.githubusercontent.com/83714022/294145127-bf9c35b3-532c-4887-84e6-eb87a18cf36e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

I think this happens because it's considered a system app. Therefore, it can't be uninstalled. Try disabling instead.

vijayvardan, 2024-01-05:
> <Universal-Debloater-Alliance/universal-android-debloater-next-generation/issues/112/1877588787
> @github.com>

I believe the system apps can be uninstalled too, Yes I did disable them!

AnonymousWP, 2024-01-07:
> I believe the system apps can be uninstalled too, Yes I did disable them!
> […](#)

Some can't be as they're on the system partition (which is not writeable by the user). Disabling should be the way to go.

closed 2024-01-07 by AnonymousWP