Universal-Debloater-Alliance/universal-android-debloater-next-generation #121: Visual bugs in selection review

Frigyes06, 2023-12-26:
**Describe the bug**
Alignments / layout in the review screen is skewed in some scenarios. For example, the category titles are clipped on the bottom, and the uninstall verb can be pushed by a long package name.

[image]

**Expected behavior**
The layout is nice and consise

**You have a solution?**
Nope, no solution yet.

**UAD log**
Upload the logfile generated by UAD.
```
2023-12-25 19:32:10 DEBUG [src/core/update.rs:174] Checking for UAD update
2023-12-25 19:32:10 INFO  [src/gui/views/list.rs:125] -----------------------------------------------------------------
2023-12-25 19:32:10 INFO  [src/gui/views/list.rs:126] ANDROID_SDK: 29 | DEVICE: HUAWEI MAR-LX1A
2023-12-25 19:32:10 INFO  [src/gui/views/list.rs:130] -----------------------------------------------------------------
2023-12-25 19:35:05 DEBUG [src/core/update.rs:174] Checking for UAD update
2023-12-25 19:37:13 WARN  [/home/frigyes/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.16.2/src/vulkan/instance.rs:565] Unable to find layer: VK_LAYER_KHRONOS_validation
2023-12-25 19:37:14 WARN  [/home/frigyes/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.16.2/src/vulkan/instance.rs:611] create_instance: ERROR_INCOMPATIBLE_DRIVER
2023-12-25 19:37:14 WARN  [/home/frigyes/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.16.2/src/gles/adapter.rs:578] Detected skylake derivative running on mesa i915. Clears to srgb textures will use manual shader clears.
2023-12-25 19:37:14 WARN  [/home/frigyes/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.16.2/src/gles/adapter.rs:578] Detected skylake derivative running on mesa i915. Clears to srgb textures will use manual shader clears.
```

AnonymousWP, 2024-03-18:
Do you think this has been solved by https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/pull/334?

Frigyes06, 2024-04-14:
[image]

Still present

PR #452 (2024-05-08): bug(ui): visual bugs in selection review
 - increased width of modal

 - adjusted some column sizes

[image]

Resolves #121
closed 2024-05-08 by adhirajsinghchauhan