Universal-Debloater-Alliance/universal-android-debloater-next-generation #129: Tool lacks a way to use a custom list (aka preselected options) to debloat

allanonmage, 2023-12-27:
**Describe the bug**
The original UAD had a process that you identified what was on your device, then compiled a list of things to remove, then you fed the tool that list.  I have my list from 2021, yet neither the original Rust version (0.5), nor this active version (0.6.2) have any way to import the list.  There's a looooooooooooooooooooooong list of checkboxes (terrible UI for large numbers of selections).  V5 would accept an override of "uad_exported_selection.txt", but it doesn't appear that yours does, based on the number of the "Review Selection" button.  On v5, that's a differently named button, and shows the number of selections, or the number of packages listed in the txt file.

The log file talks about "[src\gui\views\list.rs:321] Selection exported", but that folder path doesn't match the local folder, nor the APPDATA one, looks like a Linux path, and looks like it's part of the build path, not the runtime path.

**Expected behavior**
An import button to accept and parse a list of packages to remove.

An FAQ or guide that goes over how to use the above functionality.

**You have a solution?**
Add the expected functionality.  Is this a homework proof or concept that you would skip such an obvious requirement? Like, idk how you could ever conceive of this kind of software without such a button, so I'm going to come across as condescending regardless of word choice.  It's just a dead-simple part of such a utility that I am flabbergasted that I have to explain this to the 3rd iteration of this tool.

**UAD log**
Upload the logfile generated by UAD.
[UAD_20231227.log](https://github.com/Universal-Debloater-Alliance/universal-android-debloater/files/13781943/UAD_20231227.log)

Frigyes06, 2023-12-27:
I agree with OP, the selection export & Import function should be rewritten, hopefully better than the original implementation. This would probably be a v2 backlog item though.

allanonmage, 2023-12-27:
While I understand what you mean by saying it's a v2 thing, how does that work with a less than 0 current version number?  That saying assumes that the current version is a v1 of some kind.  Are you saying the next version, or something else?

Frigyes06, 2023-12-27:
Look at the v1 milestone, we're almost done. v2 should come when the program changes significantly, like a UI flow or functionality change. I'm just saying it won't fit in v1 backlog