Universal-Debloater-Alliance/universal-android-debloater-next-generation #130: Export functionality was removed?

allanonmage, 2023-12-27:
**Describe the bug**
There appears to be no way to export the list of selected options, like there is in v5 of the tool

**Expected behavior**
A button that makes a file called "uad_exported_selection.txt"

**You have a solution?**
Put the button back.

**UAD log**
n/a

Frigyes06, 2023-12-27:
See:
https://github.com/0x192/universal-android-debloater/issues/192

Please open a feature request

closed 2023-12-27 by Frigyes06

allanonmage, 2023-12-27:
The feature that was implemented in the linked post, in the previous version, no longer exists in the current version.  Since I was using the latest version of the other project, that means that this project has broken (possibly removed?) that functionality.  Since it was part of a button that no longer exists, idk which one it is.