Universal-Debloater-Alliance/universal-android-debloater-next-generation #134: Config file in a non-apparent place, in a 2nd APPDATA location

allanonmage, 2023-12-28:
**Describe the bug**
I was not able to get v6 to work, so I tried to use the v5.  v5 would not run because you are using a config file in a APPDATA folder.  Once I deleted the config file, v5 would launch.  Both the local and roaming APPDATA were used for different things, making things difficult to track down.  Which

**Expected behavior**
A portable software that doesn't need an installation routine should not be putting stuff anywhere, especially if it interferes with other programs. This is against 40 years of Windows conventions when software that doesn't have an installer (known as portable software) puts files all over the place in an undocumented fashion.  You know who is famous for doing that?  Norton antivirus.

**You have a solution?**
Stop using APPDATA and other random weird stuff.

**UAD log**
Upload the logfile generated by UAD.

adhirajsinghchauhan, 2023-12-28:
Duplicate of #128. Please don't create issues when there's one open for exactly the same thing.

closed 2023-12-28 by adhirajsinghchauhan

allanonmage, 2024-01-06:
No, they are not the same thing.  

"Both the **local** and **roaming** APPDATA were used for different things"

The other one is for the APPDATA in general, and this one is the fact that multiple different APPDATA folders are used.  Subtle, yet different.

I've never seen a Windows software use multiple different APPDATA folders before.  Most use only roaming.