Universal-Debloater-Alliance/universal-android-debloater-next-generation #135: v6 does not appear to work

allanonmage, 2023-12-28:
**Describe the bug**
v6 does not appear to work.  There's no manual nor a guide, the UI is not intuitive, and I was not able to figure out a way to get it to work.  v5 seems to still work, once you figure out how to use it.

**Expected behavior**
Newer, supported, software should work, especially if the older, unsupported software that it's based on works.

**You have a solution?**
I did adb manually, and also tried v5 and those worked fine.

**UAD log**
Upload the logfile generated by UAD.

adhirajsinghchauhan, 2023-12-28:
> Upload the logfile generated by UAD.

Please [attach logs](https://github.com/Universal-Debloater-Alliance/universal-android-debloater/wiki/FAQ#where-are-the-logfiles). We can't do anything without it.

prijkes, 2024-01-03:
You need to put the file in the same directory as where adb.exe is (on Windows).
After starting it, loading the necessary files take a while(?).
I.e. in my case the log show the following:

2024-01-03 14:41:06 DEBUG [src\core\update.rs:174] Checking for UAD update
2024-01-03 14:41:21 DEBUG [src\core\update.rs:174] Checking for UAD update
2024-01-03 14:41:22 ERROR [src\core\utils.rs:83] Can't open the following URL: 
2024-01-03 14:42:11 INFO  [src\gui\views\list.rs:125] -----------------------------------------------------------------
2024-01-03 14:42:11 INFO  [src\gui\views\list.rs:126] ANDROID_SDK: 0 | DEVICE: fetching devices...
2024-01-03 14:42:11 INFO  [src\gui\views\list.rs:130] -----------------------------------------------------------------
2024-01-03 14:42:11 ERROR [src\gui\views\list.rs:713] AppsView ready but no phone found

Frigyes06, 2024-01-03:
List update won't work as we changed the repo name. Will push a release later today

closed 2024-02-29 by Frigyes06