Universal-Debloater-Alliance/universal-android-debloater-next-generation #158: Removal of Simple-Mobile Tools ( Simple Keyboard and Simple Gallery ) from recommendation sections

im-notpua, 2024-01-06:
**Your phone model**: 

Any Xiaomi, Samsung, Pixel device

**Packages documentation to update:** 

```
com.samsung.android.honeyboard
com.touchtype.swiftkey
com.miui.gallery
com.google.android.inputmethod.latin
```

## Documentation Change

**List**: `Google`|`Misc`|`OEM` (manufacturer)

**Removal**:

None

### Current description

> Simple (product) is a good FOSS, lightweight replacement for system default, https://f-droid.org/en/packages/(package-link)

- The current description contains recommending certain apps from 'Simple-Mobile-Tools'.
- But as of 2024, sadly the company behind this suite of applications has been bought by an Ad-company.

Here are some related articles [1](https://www.ghacks.net/2023/12/05/warning-simple-mobile-tools-sold-to-controversial-zipoapps-publisher/), [2](https://www.androidauthority.com/simple-mobile-tools-acquisition-3391041/), [3](https://vid.puffyan.us/watch?v=OGr4H7QLVRs). 

Which would mean it can be controversial to recommend apps from these developers.

### Proposed description

Either remove these recommendations, or replace them with better ones like ( if not already present ) ['Unexpected Keyboard'](https://f-droid.org/en/packages/juloo.keyboard2), ['LeafPic-Revived'](https://f-droid.org/en/packages/com.alienpants.leafpicrevived).

Frigyes06, 2024-01-07:
Ooof. Yeah. That's sad

lavafroth, 2024-01-13:
Maybe we can add openboard or florisboard. Florisboard has glide typing like some proprietary keyboards. Also, for voice typing, there is sayboard.

AnonymousWP, 2024-01-15:
> Maybe we can add openboard or florisboard. Florisboard has glide typing like some proprietary keyboards. Also, for voice typing, there is sayboard.

OpenBoard is in the description already. I'm personally not a fan of recommending keyboards in a description. That's up to the user.

lavafroth, 2024-01-16:
Note that the F-droid versions of the Simple apps could still be recommended.

AnonymousWP, 2024-01-16:
Why? Like I said: I'm not going to make choices for UAD-ng users.

im-notpua, 2024-01-16:
> Note that the F-droid versions of the Simple apps could still be recommended.

F-droid tends to maintain forks of projects and strip away advertisements and "buy premium" pop ups. But now the project is likely being made closed source, so it wouldn't be possible to maintain it, and the current f-droid version won't receive any updates.

AnonymousWP, 2024-01-16:
Exactly. That too. But it seems like there are two 'simple' apps; one being from the Simple Mobile Tools project that has now been commercialised. The other one being named "Simple Keyboard" as well. So what app are we talking about? This is exactly why I think recommendations don't fit in descriptions and we should get rid of these kind of biased, subjective recommendations.

lavafroth, 2024-01-16:
Do take a look at [Fossify](https://github.com/FossifyOrg)

commit 843e9779 (2024-01-08): Removed all recommendations of Simple apps. Fixes #158
commit 73c97b00 (2024-01-09): Removed all recommendations of Simple apps. Fixes #158
commit 06df8676 (2024-01-15): Removed all recommendations of Simple apps. Fixes #158
PR #161 (2024-01-26): pkg(simple-apps): remove all recommendations of Simple apps
Fixes #158
closed 2024-01-26 by AnonymousWP