Universal-Debloater-Alliance/universal-android-debloater-next-generation #174: Enhance summary about 'com.sec.android.app.launcher'

Gabriel-Alves-Cunha, 2024-01-13:
**Your phone model**:
Samsung Galaxy S21 FE 5G

**Describe the issue**
Hello, 👋! First, I'd like to thank you all for this app! I feel my phone faster, the battery seems to last a little longer, and it seems to get less heated 😄!

The ability to change apps quickly by pressing the button to the side of the on-screen home button (I forget the name of this ability 🙃) stops working after uninstalling  'com.sec.android.app.launcher'. I do use another launcher, 'Nova Launcher', so I thought uninstalling it would be fine and remove over a hundred MB from RAM 🥲.

[image]

So, I believe it would be nice to enhance the little summary by warning users about it.
What do you think?

**You have a solution?**
Just press the 'Restore' button and that's it 😃!

**UAD-ng log**

> Upload the logfile generated by UAD-ng. It would be difficult to help you without it.

There's only this on the log of the date I managed to fix it:
``` text
2024-01-13 10:26:47 DEBUG [src/core/update.rs:174] Checking for UAD update
2024-01-13 10:26:47 INFO  [src/gui/views/list.rs:125] -----------------------------------------------------------------
2024-01-13 10:26:47 INFO  [src/gui/views/list.rs:126] ANDROID_SDK: 33 | DEVICE: samsung SM-G990E
2024-01-13 10:26:47 INFO  [src/gui/views/list.rs:130] -----------------------------------------------------------------
2024-01-13 10:27:29 INFO  [src/core/sync.rs:123] [Expert] cmd package install-existing --user 0 com.osp.app.signin -> Package com.osp.app.signin installed for user: 0
2024-01-13 10:27:47 INFO  [src/core/sync.rs:123] [Expert] cmd package install-existing --user 0 com.sec.android.app.launcher -> Package com.sec.android.app.launcher installed for user: 0
2024-01-13 10:31:32 INFO  [src/core/sync.rs:123] [Expert] pm uninstall --user 0 com.sec.android.app.launcher -> Success
2024-01-13 10:31:35 INFO  [src/core/sync.rs:123] [Expert] cmd package install-existing --user 0 com.sec.android.app.launcher -> Package com.sec.android.app.launcher installed for user: 0
```

AnonymousWP, 2024-02-09:
Could you propose a descriptive text? So you're saying disabling/uninstalling `com.sec.android.app.launcher` breaks Nova Launcher?

AnonymousWP, 2024-02-25:
@Gabriel-Alves-Cunha Please follow up on this, otherwise this issue will be closed.

Gabriel-Alves-Cunha, 2024-02-25:
Forgive me, I forgot about it 🙃.

> Could you propose a descriptive text?

Sure, how about: "Disabling/Uninstalling this is known to break the multitasking button on Samsung Galaxy S21 FE 5G with Nova Launcher (nothing happens when the button is pressed)."?

> So you're saying disabling/uninstalling com.sec.android.app.launcher breaks Nova Launcher?

It breaks the multitask button. It does nothing when pressed.
Everything else on Nova Launcher worked normally.

AnonymousWP, 2024-02-25:
Yep, thanks! PR has been made and will be merged ASAP. 🙂

closed 2024-02-29 by Frigyes06