Universal-Debloater-Alliance/universal-android-debloater-next-generation #175: new packagess

Skizzy24, 2024-01-13:
**Your phone model:** A53

**Packages:**

```
com.sec.android.photoremasterservice
com.samsung.android.wallpaper.res
```

- [x] **I removed all those packages on my phone**
      If not why. Leave the brackets blank and explain why.

## Document each package the best you can

**List**: `Google`|`Misc`|`OEM` (manufacturer)|`AOSP`|`Pending`|`Carrier` (isp).

**Removal**:

- Recommended -- Pointless or outright negative packages, and/or apps available through Google Play.
- Advanced -- Breaks obscure or minor parts of functionality, or apps that aren't easily enabled/installed through Settings/Google Play. This category is also used for apps that are useful (default keyboard/gallery/launcher/music app.) but that can easily be replaced by a better alternative.
- Expert -- Breaks widespread and/or important functionality, but nothing important to the basic operation of the operating system. Removing an Expert package should not bootloop the device (unless mentioned in the description) but we can't guarantee it 100%.
- Unsafe -- Can break vital parts of the operating system. Removing an Unsafe package have an extremely high risk of bootlooping your device.

### \<PhotoRemasterService\>

**List**: \<oem\>
**Removal**: \<advanced\>

> If removed photo remaster in gallery wont work.

### \<com.samsung.android.wallpaper.res\>

**List**: \<oem\>
**Removal**: \<advanced\>

> Preinstalled wallpapers.

AnonymousWP, 2024-02-08:
> com.sec.android.photoremasterservice

This must be incorrect. It's probably `com.samsung.sec.android.photoremasterservice, see https://www.google.com/search?q=com.sec.android.photoremasterservice and https://www.apkmirror.com/apk/samsung-electronics-co-ltd/photoremasterservice/.

> <com.samsung.android.wallpaper.res>
> List: <oem>
> Removal: <advanced>
> 
> Preinstalled wallpapers.

Why would wallpapers be `advanced`? You can always re-download wallpapers.

Skizzy24, 2024-02-10:
> > com.sec.android.photoremasterservice
> 
> This must be incorrect. It's probably `com.samsung.sec.android.photoremasterservice, see https://www.google.com/search?q=com.sec.android.photoremasterservice and https://www.apkmirror.com/apk/samsung-electronics-co-ltd/photoremasterservice/.
> 
> > <com.samsung.android.wallpaper.res>
> > List: 
> > Removal: 
> > Preinstalled wallpapers.
> 
> Why would wallpapers be `advanced`? You can always re-download wallpapers.

yeah i wrote it wrong: com.samsung.android.photoremasterservice
also, wallpapers are advanced because the background will be transparent(black but the system will see it as white) unless if you use a custom wallpaper, also the logcat gave errors

AnonymousWP, 2024-02-11:
Alright, thanks. Opened a PR.

closed 2024-02-12 by AnonymousWP