Universal-Debloater-Alliance/universal-android-debloater-next-generation #176: ERROR [src/core/sync.rs:129] [Recommended] pm enable com.google.android.gms.supervision -> Exception occurred while executing 'enable'

homocomputeris, 2024-01-14:
**Describe the bug**
```
ERROR [src/core/sync.rs:129] [Recommended] pm enable com.google.android.gms.supervision ->
Exception occurred while executing 'enable':
java.lang.SecurityException: Shell cannot change component state for null to 1
	at com.android.server.pm.PackageManagerService.setEnabledSettings(PackageManagerService.java:4360)
	at com.android.server.pm.PackageManagerService.-$$Nest$msetEnabledSettings(PackageManagerService.java:0)
	at com.android.server.pm.PackageManagerService$IPackageManagerImpl.setApplicationEnabledSetting(PackageManagerService.java:6424)
	at com.android.server.pm.PackageManagerShellCommand.runSetEnabledSetting(PackageManagerShellCommand.java:2537)
	at com.android.server.pm.PackageManagerShellCommand.onCommand(PackageManagerShellCommand.java:282)
	at com.android.modules.utils.BasicShellCommandHandler.exec(BasicShellCommandHandler.java:97)
	at android.os.ShellCommand.exec(ShellCommand.java:38)
	at com.android.server.pm.PackageManagerService$IPackageManagerImpl.onShellCommand(PackageManagerService.java:7084)
	at android.os.Binder.shellCommand(Binder.java:1104)
	at android.os.Binder.onTransact(Binder.java:914)
	at android.content.pm.IPackageManager$Stub.onTransact(IPackageManager.java:4905)
	at com.android.server.pm.PackageManagerService$IPackageManagerImpl.onTransact(PackageManagerService.java:7068)
	at android.os.Binder.execTransactInternal(Binder.java:1380)
	at android.os.Binder.execTransact(Binder.java:1311)
thread 'main' panicked at src/gui/views/list.rs:259:59:
index out of bounds: the len is 0 but the index is 0
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

**Expected behavior**
Do not throw an error.

**You have a solution?**
No, I don't.

**UAD-ng log**
```
2024-01-14 11:06:33 DEBUG [src/core/update.rs:174] Checking for UAD-ng update
2024-01-14 12:24:46 DEBUG [src/core/update.rs:174] Checking for UAD-ng update
2024-01-14 12:24:47 INFO  [src/gui/views/list.rs:121] -----------------------------------------------------------------
2024-01-14 12:24:47 INFO  [src/gui/views/list.rs:122] ANDROID_SDK: 34 | DEVICE: samsung SM-A346B
2024-01-14 12:24:47 INFO  [src/gui/views/list.rs:126] -----------------------------------------------------------------
2024-01-14 12:24:55 ERROR [src/core/sync.rs:129] [Recommended] pm enable com.google.android.gms.supervision -> 
Exception occurred while executing 'enable':
java.lang.SecurityException: Shell cannot change component state for null to 1
	at com.android.server.pm.PackageManagerService.setEnabledSettings(PackageManagerService.java:4360)
	at com.android.server.pm.PackageManagerService.-$$Nest$msetEnabledSettings(PackageManagerService.java:0)
	at com.android.server.pm.PackageManagerService$IPackageManagerImpl.setApplicationEnabledSetting(PackageManagerService.java:6424)
	at com.android.server.pm.PackageManagerShellCommand.runSetEnabledSetting(PackageManagerShellCommand.java:2537)
	at com.android.server.pm.PackageManagerShellCommand.onCommand(PackageManagerShellCommand.java:282)
	at com.android.modules.utils.BasicShellCommandHandler.exec(BasicShellCommandHandler.java:97)
	at android.os.ShellCommand.exec(ShellCommand.java:38)
	at com.android.server.pm.PackageManagerService$IPackageManagerImpl.onShellCommand(PackageManagerService.java:7084)
	at android.os.Binder.shellCommand(Binder.java:1104)
	at android.os.Binder.onTransact(Binder.java:914)
	at android.content.pm.IPackageManager$Stub.onTransact(IPackageManager.java:4905)
	at com.android.server.pm.PackageManagerService$IPackageManagerImpl.onTransact(PackageManagerService.java:7068)
	at android.os.Binder.execTransactInternal(Binder.java:1380)
	at android.os.Binder.execTransact(Binder.java:1311)
2024-01-14 12:25:16 INFO  [src/core/sync.rs:123] [Recommended] cmd package install-existing com.monotype.android.font.samsungone -> Package com.monotype.android.font.samsungone installed for user: 0
2024-01-14 12:25:16 INFO  [src/core/sync.rs:123] [Recommended] cmd package install-existing android.autoinstalls.config.samsung -> Package android.autoinstalls.config.samsung installed for user: 0
2024-01-14 12:25:16 INFO  [src/core/sync.rs:123] [Recommended] cmd package install-existing com.aura.oobe.samsung.gl -> Package com.aura.oobe.samsung.gl installed for user: 0
2024-01-14 12:25:17 INFO  [src/core/sync.rs:123] [Recommended] cmd package install-existing com.microsoft.skydrive -> Package com.microsoft.skydrive installed for user: 0
2024-01-14 12:25:17 INFO  [src/core/sync.rs:123] [Recommended] cmd package install-existing com.google.android.apps.tachyon -> Package com.google.android.apps.tachyon installed for user: 0
2024-01-14 12:25:18 INFO  [src/core/sync.rs:123] [Recommended] cmd package install-existing com.samsung.android.app.camera.sticker.facearavatar.preload -> Package com.samsung.android.app.camera.sticker.facearavatar.preload installed for user: 0
2024-01-14 12:25:18 INFO  [src/core/sync.rs:123] [Recommended] cmd package install-existing com.google.android.gm -> Package com.google.android.gm installed for user: 0
2024-01-14 12:25:18 INFO  [src/core/sync.rs:123] [Recommended] cmd package install-existing com.samsung.android.app.cocktailbarservice -> Package com.samsung.android.app.cocktailbarservice installed for user: 0
2024-01-14 12:25:18 INFO  [src/core/sync.rs:123] [Recommended] cmd package install-existing com.samsung.aasaservice -> Package com.samsung.aasaservice installed for user: 0
2024-01-14 12:25:18 INFO  [src/core/sync.rs:123] [Recommended] cmd package install-existing com.microsoft.appmanager -> Package com.microsoft.appmanager installed for user: 0
2024-01-14 12:25:19 INFO  [src/core/sync.rs:123] [Recommended] cmd package install-existing com.samsung.android.aircommandmanager -> Package com.samsung.android.aircommandmanager installed for user: 0
2024-01-14 12:25:19 INFO  [src/core/sync.rs:123] [Recommended] cmd package install-existing com.google.android.youtube -> Package com.google.android.youtube installed for user: 0
2024-01-14 12:25:19 INFO  [src/core/sync.rs:123] [Recommended] cmd package install-existing com.samsung.android.allshare.service.mediashare -> Package com.samsung.android.allshare.service.mediashare installed for user: 0
2024-01-14 12:25:20 INFO  [src/core/sync.rs:123] [Recommended] cmd package install-existing com.samsung.android.app.clipboardedge -> Package com.samsung.android.app.clipboardedge installed for user: 0
2024-01-14 12:25:20 INFO  [src/core/sync.rs:123] [Recommended] cmd package install-existing com.samsung.android.app.omcagent -> Package com.samsung.android.app.omcagent installed for user: 0
2024-01-14 12:25:20 INFO  [src/core/sync.rs:123] [Recommended] cmd package install-existing com.samsung.android.app.reminder -> Package com.samsung.android.app.reminder installed for user: 0
2024-01-14 12:25:21 INFO  [src/core/sync.rs:123] [Recommended] cmd package install-existing com.samsung.android.app.watchmanagerstub -> Package com.samsung.android.app.watchmanagerstub installed for user: 0
2024-01-14 12:25:21 INFO  [src/core/sync.rs:123] [Recommended] cmd package install-existing com.samsung.android.app.taskedge -> Package com.samsung.android.app.taskedge installed for user: 0
2024-01-14 12:25:21 INFO  [src/core/sync.rs:123] [Recommended] cmd package install-existing com.samsung.android.app.routines -> Package com.samsung.android.app.routines installed for user: 0
2024-01-14 12:25:22 INFO  [src/core/sync.rs:123] [Recommended] cmd package install-existing com.samsung.android.app.settings.bixby -> Package com.samsung.android.app.settings.bixby installed for user: 0
2024-01-14 12:25:22 INFO  [src/core/sync.rs:123] [Recommended] cmd package install-existing com.samsung.android.ardrawing -> Package com.samsung.android.ardrawing installed for user: 0
2024-01-14 12:25:23 INFO  [src/core/sync.rs:123] [Recommended] cmd package install-existing com.samsung.android.aremoji -> Package com.samsung.android.aremoji installed for user: 0
2024-01-14 12:25:23 INFO  [src/core/sync.rs:123] [Recommended] cmd package install-existing com.samsung.android.authfw -> Package com.samsung.android.authfw installed for user: 0
2024-01-14 12:25:24 INFO  [src/core/sync.rs:123] [Recommended] cmd package install-existing com.samsung.android.bbc.bbcagent -> Package com.samsung.android.bbc.bbcagent installed for user: 0
[... cut ...]
2024-01-14 12:25:26 INFO  [src/core/sync.rs:123] [Recommended] cmd package install-existing com.samsung.android.easysetup -> Package com.samsung.android.easysetup installed for user: 0
2024-01-14 12:25:26 INFO  [src/core/sync.rs:123] [Recommended] cmd package install-existing com.samsung.android.fast -> Package com.samsung.android.fast installed for user: 0
2024-01-14 12:25:26 INFO  [src/core/sync.rs:123] [Recommended] cmd package install-existing com.samsung.android.kidsinstaller -> Package com.samsung.android.kidsinstaller installed for user: 0
```

Frigyes06, 2024-01-15:
That appears to be an exception in ADB. Is it reproducible?

homocomputeris, 2024-01-16:
For me it happens every time I check this checkbox.

AnonymousWP, 2024-02-15:
Could it be related to https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/pull/261 ?

homocomputeris, 2024-02-15:
Why does the shell thinks that `com.google.android.gms.supervision` state is `null`? If it were null, why would appear in the app list?

AnonymousWP, 2024-02-18:
Try again with our upcoming release: 1.0.1.

AnonymousWP, 2024-03-18:
@homocomputeris Follow up on this.

AnonymousWP, 2024-03-24:
Closing as user did not follow up.

closed 2024-03-24 by AnonymousWP