Universal-Debloater-Alliance/universal-android-debloater-next-generation #182: dark theme or light theme?

bar81max, 2024-01-20:
just sync it with the os 🥴

AnonymousWP, 2024-01-20:
This should be an option; not the default.

Rudxain, 2024-07-04:
I'm willing to contribute to this. However, I'm still trying to get familiarized with the codebase.

[`dark-light` doesn't provide `async`](https://docs.rs/dark-light/1.1.1/dark_light/#functions) API, so color-scheme can only be efficiently synced on app startup. We could use a polling mechanism to update in real-time, but that seems overkill, since most users won't frequently change the DE theme while UAD is running.

In the future, we could copy some code from Firefox (is `MPL-2.0` compatible with `GPL-3.0`?), to update in real-time in an event-driven way

AnonymousWP, 2024-07-04:
It's fine if it's only on start-up. If it's not too much work it's a nice bonus, but not mandatory for sure.🙂

Rudxain, 2024-10-27:
#650 #627

PR #531 (2024-07-10): feat(theme): follow system theme by default
closes #182

ℹ️ https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/issues/182#issuecomment-2208347197
closed 2024-11-03 by adhirajsinghchauhan