Universal-Debloater-Alliance/universal-android-debloater-next-generation #188: infinite error logs if adb isn't installed

jxdv, 2024-01-26:
**Describe the bug**
If adb isn't installed than 2 errors are logged infinitely. (UAD-ng version 0.6.2.)
[image]

**Expected behavior**
If adb isn't installed than logs the 2 error once and exit with error code 1.

**You have a solution?**
Exit right away if adb isn't installed.

AnonymousWP, 2024-01-30:
One of our devs should look into this. Or, if you can help us out, file a PR.

jxdv, 2024-01-31:
> One of our devs should look into this. Or, if you can help us out, file a PR.

I am not familiar with Rust, so I'd rather not do it myself, but it looks like an easy fix of exiting the process with code 1 if this error gets caught.
```rust
match command.output() {
    Err(e) => {
        error!("ADB: {}", e);
        Err("ADB was not found".to_string())
}

PR #198 (2024-02-01): fix: exit when adb is not found
Fixes: #188

The program prematurely exits with a status code of 1 when ADB is not found.
closed 2024-02-01 by adhirajsinghchauhan