Universal-Debloater-Alliance/universal-android-debloater-next-generation #190: 1 persistent nonexisting package in review selection window

prakharb5, 2024-01-26:
**Describe the bug**

[image]

I have only selected one package. But it is counted twice in 2 separate categories.
The selected package is from Recommended category.

[image]

Selecting one package in Advanced category shows up as 2 packages in Advanced category.

**Expected behavior**
No error in enumeration

**You have a solution?**
NA

**UAD-ng log**
Upload the logfile generated by UAD-ng.

prakharb5, 2024-01-27:
Restarting the app somehow fixed it, and now I cannot reproduce this bug :/

AnonymousWP, 2024-01-29:
Add a comment in this issue if you can reproduce it, then I'll re-open it. Closing it for now.

closed 2024-01-29 by AnonymousWP