Universal-Debloater-Alliance/universal-android-debloater-next-generation #193: Additional for realme (and certain Thai baked-in)

kowith337, 2024-01-27:
**Your phone model:** realme 11x (RMX3785)

**Packages:**
| PackageName | App Name | Reason |
| - | - | - |
| `cn.wps.moffice_eng` | `WPS Office` | WPS is a Chinese closed-source software. It's as bad as Microsoft Office (privacy-wise) |
| `com.agoda.mobile.consumer` | `Agoda` | Unneeded if you didn't need any deals and/or offers to reserve rooms during travel. |
| `com.booking` | `Booking` | Unneeded if you're not travel too much, I'm also heard that `Huawei` also load wallpapers from that app/site just for advertising. |
| `com.igg.android.lordsmobile` | `Lords Mobile` | Preloaded Game, you can manually install later if you want to... |
| `com.lazada.android` | `Lazada` | Online shopping apps are often promote in my region (as malicious way, e.g. inject web notifications, force open play store to download, etc.), some are shift with common branded devices that sales on Thai, even Hi-end `Samsung`, also those kind of apps are often request suspicious permissions too! |
| `com.linkedin.android` | `LinkedIn` | Not necessary, also I heard about privacy breach incident earlier. |
| `com.miHoYo.GenshinImpact` | `Genshin Impact` | Preloaded Game, you can manually install later if you want to... |
| `com.miHoYo.bh3oversea` | `Honkai Impact` | Preloaded Game, you can manually install later if you want to... |
| `com.shopee.th` | `Shopee` (TH) | Same reason as Lazada |
| `com.spotify.music` | `Spotify` | Preloaded (and untrusted for privacy-wise) music streaming app, you can manually install later if you want to... |
| `com.ss.android.ugc.trill` | `TikTok` | Privacy Aware (you know, right?) |

- [x] **I removed all those packages on my phone**

## Document each package the best you can

**List**: All are Misc (probably, maybe pending?)

**Removal**: All must be labeled as **Recommend**

kowith337, 2024-01-30:
Edited because those apps are semi-preload due to I've test with AppManager and doesn't found it as removed system apps, semi-preload is mean apps will came back and install as user apps even perform wipe entire phone, but it doesn't count as part of system.

AnonymousWP, 2024-02-17:
Sorry for the late reply and thank you for this detailed issue.

I know all of those apps, but they're pre-installed? Also, when someone has those applications downloaded themselves, UAD-ng will recommend them to uninstall. Isn't that inconvenient? I think that's what you mean, but maybe Google back-up caused the apps to being installed again? That's usually the case when you want to restore a back-up or whatsoever.

kowith337, 2024-02-18:
It's preinstalled but it doesn't look like it's binding as system apps.
I'm always won't let devices use any internet, no matter about Wi-Fi and/or mobile because I didn't put sim card into devices at first and always do that whenever I've got new devices.
That's mean it's not possible to tell that it comes with Google backup or anything.
Whenever devices are wiped, those apps still come back, so assumed that it's come with regional-specific ROM as semi-preload.

AnonymousWP, 2024-02-25:
@Frigyes06 Can I have your opinion on this? Should we treat this as system app or not? As they're available in the Play Store as well.

AnonymousWP, 2024-03-11:
@Frigyes06 Bump.

AnonymousWP, 2024-03-24:
@kowith337 Can you remove those apps by hand perhaps? As in: can't you just uninstall/disable them via settings in Android?

kowith337, 2024-03-24:
All of them can be uninstallable, however, since they're doesn't actually tied as part of system apps, all of them will be gone and unrecoverable until perform phone wipe next time (or caused bootloop and need to wipe phone for reset)
Note that this device are contains Facebook main app as stub version, probably it will download the main version if open its app for first time, well, it's come with manager and services as system apps, that's mean Facebook app will have enough privileges to self-update because of those, I think...

schbrns, 2024-11-24:
I think there is a distinction between **system apps** and _pre-installed user apps_. 
In the case of Facebook, that has been and still is a system app on some devices. 
More importantly, it can be uninstalled by the user directly on-device without needing to enable additional options.

Feel free to re-open the issue if you feel it hasn't been resolved.
Also, should these apps ever appear as system apps, feel free to reopen the issue.

closed 2024-11-24 by schbrns