Universal-Debloater-Alliance/universal-android-debloater-next-generation #200: Suggestion on packages

xe1st, 2024-02-01:
**Your phone model**:
Poco x3 nfc

**Packages**
```com.android.providers.downloads.ui```
recommended change: Move to *recommended* category from *unsafe*. Just an app that shows downloaded files, it is not necessary at all.

```com.android.soundrecorder```
recommended change: Move to *recommended* category from *advanced*(iirc). Voice recorder and easily replacable.

✅ I uninstalled them on my phone

AnonymousWP, 2024-02-04:
> com.android.providers.downloads.ui
recommended change: Move to recommended category from unsafe. Just an app that shows downloaded files, it is not necessary at all.

The question is: can you easily download & set another one that you can set as default download application?

xe1st, 2024-02-04:
> > com.android.providers.downloads.ui
> > recommended change: Move to recommended category from unsafe. Just an app that shows downloaded files, it is not necessary at all.
> 
> The question is: can you easily download & set another one that you can set as default download application?

it is just a seperate gui app to see your browser downloads. You can still see your downloads from browser or xiaomi's file manager etc. Idk why there is a discrete app for that lol. Think of it like a windows application whose point is only show what your download. It doesn't control the downloads themselves.

AnonymousWP, 2024-02-06:
> it is just a seperate gui app to see your browser downloads. You can still see your downloads from browser or xiaomi's file manager etc. Idk why there is a discrete app for that lol. Think of it [...]

Yeah I know, but can you re-download it from the Play Store or something? See https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/wiki/FAQ#how-are-the-recommendations-chosen.

xe1st, 2024-02-10:
> Yeah I know, but can you re-download it from the Play Store or something? See [...]

nope not listed on playstore. I just considered it pointless tho

AnonymousWP, 2024-02-11:
I'll make it `Advanced` instead as a middle-way. Thanks.

closed 2024-02-12 by AnonymousWP