Universal-Debloater-Alliance/universal-android-debloater-next-generation #220: bug(adb): UAD-ng doesn't start when ADB isn't installed

AnonymousWP, 2024-02-07:
**Describe the bug**

When ADB is absent on a system, UAD-ng doesn't start at all. It also doesn't tell me ADB isn't installed.

**Expected behavior**

I expect UAD-ng to start and inform me that ADB is required but isn't present/installed.

**You have a solution?**

Implement a check when starting UAD-ng.

**UAD-ng log**

> 2024-02-07 10:21:35 ERROR [src\core\config.rs:79] Failed to read config file: `System cannot find the specified file. (os error 2)`
> 2024-02-07 10:21:35 ERROR [src\core\config.rs:81] Restoring default config file
> 2024-02-07 10:21:35 ERROR [src\core\sync.rs:72] ADB: program not found
> 2024-02-07 10:21:35 ERROR [src\core\sync.rs:73] ADB was not found
> 2024-02-07 10:21:59 ERROR [src\core\sync.rs:72] ADB: program not found
> 2024-02-07 10:21:59 ERROR [src\core\sync.rs:73] ADB was not found
> 2024-02-07 10:22:31 ERROR [src\core\sync.rs:72] ADB: program not found
> 2024-02-07 10:22:31 ERROR [src\core\sync.rs:73] ADB was not found

Frigyes06, 2024-02-07:
see #198

AnonymousWP, 2024-02-07:
> see #198

Yeah but my point is that we need something more verbose.

AnonymousWP, 2024-03-14:
It's not just about being verbose. UAD-ng just doesn't start at all and the user is not notified by it, unless the user checks logs. That's a very user-unfriendly way.

AnonymousWP, 2024-03-14:
Would something like this work?

```Rust
match command.output() {
    Err(e) => {
        // Log the error for debugging purposes
        error!("ADB: {}", e);
        error!("ADB was not found");

        // Display error message to the user using a message box
        message_box(None, "Error", "ADB was not found. Please make sure ADB is installed and accessible.", MessageBoxButtons::Ok, MessageBoxIcon::Error);

        // Exit the program
        std::process::exit(1);
    }
    // Handle Ok arm here if needed
}
```

closed 2024-03-17 by adhirajsinghchauhan