Universal-Debloater-Alliance/universal-android-debloater-next-generation #223: bug(uad): logo not visible on Windows 11

AnonymousWP, 2024-02-07:
**Describe the bug**

Due to the white text, logo isn't visible.

**Expected behavior**

The logo should be visible.

**You have a solution?**

No, but probably use a different logo depending on the user's theme. Like: follow system theme. If dark, use transparent logo, if not, use logo with black background.

**Screenshot**

[image]

AnonymousWP, 2024-03-02:
Would this work?

```Rust
use std::process::Command;

fn get_windows_theme() -> Option<String> {
    let output = Command::new("reg")
        .args(&["query", "HKCU\\Software\\Microsoft\\Windows\\CurrentVersion\\Themes\\Personalize", "/v", "AppsUseLightTheme"])
        .output()
        .ok()?;
    
    let output_str = String::from_utf8(output.stdout).ok()?;
    let theme_value = output_str.lines().nth(2)?;

    if theme_value.contains("0x1") {
        Some("Dark".to_string())
    } else {
        Some("Light".to_string())
    }
}

fn main() {
    match get_windows_theme() {
        Some(theme) => {
            if theme == "Dark" {
                println!("Using logo A for dark mode");
                // Display logo A
            } else {
                println!("Using logo B for light mode");
                // Display logo B
            }
        }
        None => println!("Failed to determine Windows theme."),
    }
}
```

AnonymousWP, 2024-03-14:
@Universal-Debloater-Alliance/developers Can I have your opinion on this?

lavafroth, 2024-03-15:
There's a platform independent crate called light-dark, better use that instead of reg queries lest some antivirus incorrectly flags us.

AnonymousWP, 2024-03-15:
Yeah, valid point, I thought the same. Unfortunately ChatGPT did not consider that, lol. Unfortunately I don't have Rust-knowledge, so hopefully ChatGPT knows about that crate.

closed 2024-03-23 by Frigyes06