Universal-Debloater-Alliance/universal-android-debloater-next-generation #224: bug(update-check): incorrectly reporting source/parent version

AnonymousWP, 2024-02-07:
**Describe the bug**

When opening UAD-ng, it reports that there's a new version available while there is not.

**Expected behavior**

I expect it to not return a false positive. 🙂

**You have a solution?**

Probably a coding mistake in the `update.rs` file.

**Screenshot**

[image]

edgy-b, 2024-02-09:
This error is present, because in `tag_name` besides version `v` is added to it.
https://api.github.com/repositories/710425233/releases
Possible solutions would be to change tag_name or string replace in `get_latest_release` function
```
let release_version = release.tag_name.as_str().replace("v", "");
if release_version != "dev-build"
    && release_version > env!("CARGO_PKG_VERSION").to_string()
{
    Ok(Some(release))
 } else {
    Ok(None)
}
```

AnonymousWP, 2024-02-09:
> This error is present, because in `tag_name` besides version number `v` is added to it. https://api.github.com/repositories/710425233/releases Possible solutions would be to
> 
> 1. change tag_name (easiest, without affecting code or adding extra checks)
> 2. string replace in `get_latest_release` function
> 
> ```
> let release_version = release.tag_name.as_str().replace("v", "");
> if release_version != "dev-build"
>     && release_version > env!("CARGO_PKG_VERSION").to_string()
> {
>     Ok(Some(release))
>  } else {
>     Ok(None)
> }
> ```

Thanks. @Universal-Debloater-Alliance/developers See solution proposal above. Developers will discuss with you what's the best and most viable solution.

Frigyes06, 2024-02-09:
The second solution would be "proper", if we want to follow good practices on version tagging. 
Also the other cause of the problem was that the version did not get bumped in cargo.toml
Maybe we can add a check in the release workflow to see if the version in the manifest matches the git tag, or even possibly setting it based on the git tag automatically.

PR #236 (2024-02-12): fix: strip `v` version prefix when checking for updates
Fixes #224

### Changes
- The `v` prefix from the version name is ignored if it exists. Should prevent erroneous update version.
closed 2024-02-12 by adhirajsinghchauhan