Universal-Debloater-Alliance/universal-android-debloater-next-generation #225: bug(update-check): downloading new update via GUI doesn't do anything

AnonymousWP, 2024-02-07:
**Describe the bug**

When using the built-in updater, it says it's downloading the update, but in fact nothing is happening.

**Expected behavior**

I expect it to be successful and if not, notify me what went wrong.

**You have a solution?**

No.

**UAD-ng log**

It's just stuck on

`2024-02-03 12:49:54 DEBUG [src\core\update.rs:174] Checking for UAD update`

Frigyes06, 2024-02-08:
CI is broken, making bad files, not an issue with the actual program

AnonymousWP, 2024-02-11:
> CI is broken, making bad files, not an issue with the actual program

Yeah, but then I expect it to return an error or message at least. This is too vague and it needs to be verbose.

PR #372 (2024-03-24): fix(update-check): when downloading an update failed, UAD-ng isn't verbose about it
 - removed todo in sync.rs, ADB verbose message is implemented

 - added message to display if update failed with button to go back

 - added button for ADB failed view with link to wiki page

 - tweaked waiting_view fn

resolves #225
closed 2024-03-24 by Frigyes06