Universal-Debloater-Alliance/universal-android-debloater-next-generation #239: bug(macos): can not update to v1.0.0

haunt98, 2024-02-10:
**Describe the bug**
I download macos version `uad-ng-macos` from https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/releases/tag/v1.0.0

**Expected behavior**
Shows v1.0.0

**You have a solution?**
Sadly no

**UAD-ng log**

```txt
2024-02-10 21:08:37 ERROR [src/core/config.rs:77] Invalid config file: `TOML parse error at line 1, column 1
  |
1 | theme = "Lupin"
  | ^^^^^^^^^^^^^^^
missing field `general`
`
2024-02-10 21:08:37 ERROR [src/core/config.rs:81] Restoring default config file
2024-02-10 21:08:42 DEBUG [src/core/update.rs:174] Checking for UAD-ng update
2024-02-10 21:08:49 DEBUG [src/gui/mod.rs:271] UAD-ng update has been download!
2024-02-10 21:08:49 ERROR [src/gui/mod.rs:310] Failed to update UAD-ng!
2024-02-10 21:09:26 DEBUG [src/gui/mod.rs:271] UAD-ng update has been download!
2024-02-10 21:09:26 ERROR [src/gui/mod.rs:310] Failed to update UAD-ng!
2024-02-10 21:09:38 DEBUG [src/core/update.rs:174] Checking for UAD-ng update
2024-02-10 21:09:41 DEBUG [src/gui/mod.rs:271] UAD-ng update has been download!
2024-02-10 21:09:41 ERROR [src/gui/mod.rs:310] Failed to update UAD-ng!
2024-02-10 21:10:22 DEBUG [src/core/update.rs:174] Checking for UAD-ng update
2024-02-10 21:10:30 DEBUG [src/core/update.rs:174] Checking for UAD-ng update
2024-02-10 21:10:36 INFO  [src/gui/views/list.rs:137] -----------------------------------------------------------------
2024-02-10 21:10:36 INFO  [src/gui/views/list.rs:138] ANDROID_SDK: 0 | DEVICE: fetching devices...
2024-02-10 21:10:36 INFO  [src/gui/views/list.rs:142] -----------------------------------------------------------------
2024-02-10 21:10:36 ERROR [src/gui/views/list.rs:722] AppsView ready but no phone found
2024-02-10 21:10:46 DEBUG [src/core/update.rs:174] Checking for UAD-ng update
2024-02-10 21:10:48 DEBUG [src/gui/mod.rs:271] UAD-ng update has been download!
2024-02-10 21:10:48 ERROR [src/gui/mod.rs:310] Failed to update UAD-ng!
2024-02-10 21:11:52 INFO  [src/gui/views/list.rs:137] -----------------------------------------------------------------
2024-02-10 21:11:52 INFO  [src/gui/views/list.rs:138] ANDROID_SDK: 0 | DEVICE: fetching devices...
2024-02-10 21:11:52 INFO  [src/gui/views/list.rs:142] -----------------------------------------------------------------
2024-02-10 21:11:52 ERROR [src/gui/views/list.rs:722] AppsView ready but no phone found
2024-02-10 21:12:58 DEBUG [src/core/update.rs:174] Checking for UAD-ng update
2024-02-10 21:14:04 INFO  [src/gui/views/list.rs:137] -----------------------------------------------------------------
2024-02-10 21:14:04 INFO  [src/gui/views/list.rs:138] ANDROID_SDK: 0 | DEVICE: fetching devices...
2024-02-10 21:14:04 INFO  [src/gui/views/list.rs:142] -----------------------------------------------------------------
2024-02-10 21:14:05 ERROR [src/gui/views/list.rs:722] AppsView ready but no phone found
2024-02-10 21:14:27 DEBUG [src/core/update.rs:174] Checking for UAD-ng update
```

[image]

xstable, 2024-02-12:
Same issue, with other log-results. 
Started the application on Bash and got this: 

```bash
./uad-ng-linux 
Xlib:  extension "NV-GLX" missing on display ":0".
MESA-INTEL: warning: Ivy Bridge Vulkan support is incomplete
2024-02-12 10:30:45 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/fontdb-0.14.1/src/lib.rs:241] Failed to load a font face 0 from '/home/suther/.fonts/Windows/xfiles.ttf' cause font doesn't have a family name.
2024-02-12 10:30:45 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/fontdb-0.14.1/src/lib.rs:241] Failed to load a font face 0 from '/home/suther/.fonts/googleWebfonts/Bangers-Regualr.ttf' cause malformed font.
2024-02-12 10:30:45 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/fontdb-0.14.1/src/lib.rs:241] Failed to load a font face 0 from '/home/suther/.fonts/d/din1451e.ttf' cause font doesn't have a family name.
2024-02-12 10:30:45 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/fontdb-0.14.1/src/lib.rs:241] Failed to load a font face 0 from '/home/suther/.fonts/d/din1451m.ttf' cause font doesn't have a family name.
thread '<unnamed>' panicked at src/core/uad_lists.rs:197:60:
Unable to parse: Error("expected `,` or `}`", line: 4946, column: 130)
```
and if I click the Update-button in the application, this occur:
```bash
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
2024-02-12 10:31:00 ERROR [src/core/update.rs:116] Couldn't extract UAD-ng release tarball
2024-02-12 10:31:00 ERROR [src/gui/mod.rs:310] Failed to update UAD-ng!
```

Frigyes06, 2024-02-13:
Duplicate of #225

closed 2024-02-13 by AnonymousWP