Universal-Debloater-Alliance/universal-android-debloater-next-generation #242: new samsung packages

Skizzy24, 2024-02-11:
**A51 and A53**

**Packages:**

```
com.samsung.android.aliveprivacy
com.samsung.android.alive.service
com.android.intentresolver

```

- [x] **I removed all those packages on my phone**

**List**: `OEM`

**Removal**: `Recommended`.

### \<Content Suggestions\>

**List**: \<OEM\>
**Removal**: \<Recommended\>

> Content Suggestions is an AI-based app inside the Secure Folder, and its on-device AI-powered engine automatically suggests users to move private images of pre-selected categories to the Secure Folder. For this to happen users have to simply select specific faces or a type of image they want to tag as private and keep them secured in the private gallery of the Secure Folder. Once the initial setup is complete, the AI engine kicks in and identifies relevant images from the entire gallery. As this is an on-device AI solution, no information or image ever leaves the smartphone.

### \<Content Suggestions service\>

**List**: \<OEM\>
**Removal**: \<Recommended\>

> Content suggestions service.

### \<Content Suggestions service\>

**List**: \<AOSP\>
**Removal**: \<Recommended\>

> Expert
In Android 14 they moved the share menu into this package; if you uninstall it you cant longer use it.

closed 2024-02-25 by AnonymousWP

Kreuger, 2024-08-03:
Hello. I have an A54. When I upgraded to One UI 6.1, it broke the intentresolver. I have noticed here that it is a package that is recommended for removal, but a comment says this will break the package. Does anyone know how to possibly repair it? Im trying to avoid having to do a factory reset.

AnonymousWP, 2024-09-05:
> Hello. I have an A54. When I upgraded to One UI 6.1, it broke the intentresolver. I have noticed here that it is a package that is recommended for removal, but a comment says this will break the package. Does anyone know how to possibly repair it? Im trying to avoid having to do a factory reset.

Did you uninstall or disable? Also, which package is it about? Can you please submit an issue?