Universal-Debloater-Alliance/universal-android-debloater-next-generation #246: pkg Consistency

ThatGreenBlur, 2024-02-12:
**Packages documentation to update:**

```
com.google.android.configupdater
com.android.dreams.phototable     
com.google.android.apps.messaging
com.google.android.overlay.gmsconfig.gsa
com.google.android.partnersetup
com.google.mainline.telemetry
com.transsion.faceid
com.transsion.magicfont
com.transsion.repaircard
com.xui.xhide
com.transsion.applock
com.sh.smart.caller
com.transsion.calculator
com.transsion.dtsaudio
com.transsion.notebook
com.transsion.scanningrecharger
com.transsion.soundrecorder
com.sprd.linkturbo
...
```

- Advanced -- "Breaks obscure or minor parts of functionality" keep this in mind

### com.google.android.configupdater

> "Might be fine to disable if you dont use google services, Disabling might mess with security" Recommended to Advanced or maybe even Expert

### com.android.dreams.phototable    

> "See com.android.dreams.basic" com.android.dreams.basic is also Advanced, for consistency sake set it Recommended to Advanced

### com.google.android.apps.messaging

> "Removing this may cause issues with receiving 2FA" Recommended to Advanced

### com.google.android.overlay.gmsconfig.gsa  

> "safe to remove if removed com.google.android.googlequicksearchbox" com.google.android.googlequicksearchbox is also Advanced, for consistency sake set it Recommended to Advanced

### com.google.android.partnersetup

> The app apparently "enables applications to interact with google apps" thus breaking that functionality if disabled right? Recommended to Advanced

### com.google.mainline.telemetry

> "Google play uses this to determine if updates are available for the modules and to show which security patch is installed" thus breaking functionality of updates right? Recommended to Advanced

### com.transsion.faceid

> "needed for face unlock lock screen" breaks functionality of face unlock, which is something used for security reasons obviously Recommended to Advanced

### com.transsion.magicfont

> For transsion devices im pretty sure this is the only way you can install fonts, even Zfont uses this app to install fonts, breaks or well removes functionality of changing fonts if removed, Recommended to Advanced? or maybe Expert to be consistent with com.transsion.os.typeface

### com.transsion.repaircard

> the description is entirely "It's another things for China!" uhh what? so what???? it doesnt even really do anything than give you like a popup of warranty only started by Carlcare (i think), needs more information

### com.xui.xhide + com.transsion.applock

> What these apps does is basically provide the only way to hide files and lock apps in transsion launchers (Which the transsion devices require for certain functionality like multiapps not working with other launchers)  These are the only apps that hide files and lock apps with them, so shouldnt it Recommended to Advanced?

### com.sh.smart.caller +com.transsion.calculator +com.transsion.notebook +com.transsion.scanningrecharger + com.transsion.soundrecorder

> Like the google apps in Recommended, these apps have telemetry on them and are completely replaceable
> That said, com.transsion.camera & com.transsion.deskclock & com.transsion.fmradio cannot be said the same, this is because Camera is required in the lock screen, Clock is kept in background if you have a alarm unlike google's clock and fm radio is simply irreplaceable to my knowledge

### com.transsion.dtsaudio

> very safe to disable and just serves as a way to modify audio playing and *runs in the background and has telemetry* can be replaced by other better apps, Advanced to Recommended

### com.sprd.linkturbo

> "For WiFi speed?" I ran a speed test with it enabled and with it disabled, i dont see anything differentiate if not my internet is slightly better now along with other apps being disabled, Expert to Recommended

Skizzy24, 2024-02-12:
com.android.dreams.basic and com.android.dreams.phototable should be recommended, they are useless

AnonymousWP, 2024-02-17:
Daydream isn't outright negative nor can be it redownloaded from the Play Store, hence `Advanced`. Please stick to the FAQ: https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/wiki/FAQ#how-are-the-recommendations-chosen.

AnonymousWP, 2024-02-17:
> com.xui.xhide + com.transsion.applock

You can just download an app hider from the Play Store, no?

AnonymousWP, 2024-02-17:
The packages for Transsion have been added by @Blitz517 a few months ago.

AnonymousWP, 2024-02-17:
Opened a PR, see above.

ThatGreenBlur, 2024-02-18:
> The packages for Transsion have been added by @Blitz517 a few months ago.
Yes, I'm here to improve the descriptions. 
> com.xui.xhide + com.transsion.applock  You can just download an app hider from the Play Store, no?
Nope, propietary launcher, its crazy.

ThatGreenBlur, 2024-02-18:
> Opened a PR, see above.

Looks good!

closed 2024-02-18 by AnonymousWP