Universal-Debloater-Alliance/universal-android-debloater-next-generation #250: Virustotal

SFAZ38, 2024-02-12:
Hello. Is this false positive? Because one of scanners found trojan in the executable.
[image]

xe1st, 2024-02-12:
yes, false positive

AnonymousWP, 2024-02-13:
Yes it is, likely because it pulls a file and/or the application isn't signed.

closed 2024-02-13 by SFAZ38

xe1st, 2024-02-18:
bruh now windows defender does it as well
[image]

AnonymousWP, 2024-02-18:
Defender doesn't do anything here when I start it.

xe1st, 2024-02-18:
> Defender doesn't do anything here when I start it.

maybe my defender configuration is diff or something?
cause it shows the uad exe 
[image]

AnonymousWP, 2024-02-18:
Which version?

xe1st, 2024-02-18:
> Which version?

uad v1 (last version)
windows 10 ltsc 21h2

AnonymousWP, 2024-02-18:
I just updated my Defender and still nothing. Windows 11 though. But rest assured: code is reviewed by multiple users and I think it's because Defender thinks UAD-ng is downloading potential malware from an external source, while in fact it's trying to download the UAD-ng list.

xe1st, 2024-02-18:
yeah ik the app itself isn't virus but let's hope defender detection isn't common. This still wouldn't be cool for the app.