Universal-Debloater-Alliance/universal-android-debloater-next-generation #262: Version 0.6.2 instead of version 1.0.0 for Linux.

geoderix, 2024-02-15:
I downloaded version 1.0.0 for Linux - uad-ng-linux. I launched the program and it wrote that this is version 0.6.2. The update to version 1.0.0 began, which ended in a failure.
Do you check the programs that you publish as a release?
Do you know how to use Linux? Is the program generated automatically without verification, exclusively for Ubuntu?

homocomputeris, 2024-02-15:
Probably just this: https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/blob/9961bdc7a24c975be104558033bd37485bf2a944/Cargo.toml#L4C1-L4C18

homocomputeris, 2024-02-15:
I have the same issue when installing using cargo from git master.

Frigyes06, 2024-02-18:
> I have the same issue when installing using cargo from git master.

Bump the version in cargo.toml
It should work after that

geoderix, 2024-02-18:
Nothing works.
Nothing has changed.

Log:

```
2024-02-18 22:06:02 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/winit-0.28.7/src/platform_impl/linux/x11/util/randr.rs:23] XRandR reported that the display's 0mm in size, which is certifiably insane
2024-02-18 22:06:03 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.16.2/src/gles/egl.rs:226] No config found!
2024-02-18 22:06:03 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.16.2/src/gles/egl.rs:226] No config found!
2024-02-18 22:06:03 DEBUG [src/core/update.rs:174] Checking for UAD-ng update
2024-02-18 22:06:12 DEBUG [src/core/update.rs:57] downloading file from https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/releases/download/v1.0.0/uad-ng-linux.tar.gz
2024-02-18 22:06:13 ERROR [src/core/update.rs:116] Couldn't extract UAD-ng release tarball
2024-02-18 22:06:13 DEBUG [src/gui/mod.rs:271] UAD-ng update has been download!
2024-02-18 22:06:13 ERROR [src/gui/mod.rs:310] Failed to update UAD-ng!
```

Frigyes06, 2024-02-18:
> Nothing works.
> Nothing has changed.
>
> Log:
>
> ```
> 2024-02-18 22:06:02 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/winit-0.28.7/src/platform_impl/linux/x11/util/randr.rs:23] [...]

Are you trying to build latest version of the main branch?
After bumping version in cargo.toml to 1.0.0 it shouldn't even try to update.

geoderix, 2024-02-18:
Should I compile the program myself?
Wow!
Why then release under the proud name 1.0.0?

Frigyes06, 2024-02-19:
> Should I compile the program myself?
> Wow!
> Why then release under the proud name 1.0.0?

Sorry, thought you were the person who was installing from source.
Yes, v1.0.0 release is messed up, do not use it. You can wait for the patch to come out, or build from source. 
We're all humans, mistakes happen. Don't be a dick about it.

Nojevah, 2024-02-19:
How can you speak like that ? Do you really think they will want to fix it if you speak like that ?
What are you waiting to compile, share and improve it ? We will watch your commits with great interest.

geoderix, 2024-02-19:
It's funny how you expect praise for doing a bad job and misleading people. Deception in the version number, the update does not work, it does not run on Debian 11, it was compiled automatically by Github and no one checked the result. Why should I praise you? Because you promised one thing and did something completely different?
Political correctness leads to sad results. You can’t tell a white thing that it’s black.
Good luck!

adhirajsinghchauhan, 2024-02-19:
> It's funny how you expect praise for doing a bad job and misleading people. […] Why should I praise you?

@geoderix Nobody asked for praise or any of its synonyms. Deflecting like this leads nowhere. Workarounds have been presented to you, now it's up to you whether to apply them.

Your point has been made; no need for aggravation. Repeatedly hammering a nail only breaks the board.

closed 2024-02-19 by Frigyes06