Universal-Debloater-Alliance/universal-android-debloater-next-generation #270: checking sha256 won't work automatically because of file path

jxdv, 2024-02-17:
Most software which is shipped with checksums to verify that it hasn't been tampered with can be verified with these steps (on debian-based distros, but similar for different ones):
1. Download binary `imaginary-bin` and `imaginary-bin-sha256.txt`
2. sha256sum imaginary-bin
3. sha256sum --check imaginary-bin-sha256.txt

Which will result in `imaginary-bin: OK` if sums match, but won't work for uad-ng. 

1. sha256sum uad-ng-linux
2. sha256sum --check uad-ng-linux-checksum

Gives this output:
```
sha256sum: bin/uad-ng-linux: No such file or directory
bin/uad-ng-linux: FAILED open or read
sha256sum: WARNING: 1 listed file could not be read
```

Most likely because `uad-ng-linux-checksum` has following structure:

```
<hash> bin/uad-ng-linux
```

File paths don't match as we can see: `uad-ng-linux` != `bin/uad-ng-linux`. Of course It can still be verified, but it's annoying not being able to use the in-built method..

Fixing this is easy (just editing the CI). Feel free to assign me and I can do it in the upcoming days.

Frigyes06, 2024-02-19:
It'd be appreciated if you could implement the fix

jxdv, 2024-02-20:
> It'd be appreciated if you could implement the fix

Sure, I'll take a look in the upcoming days.

PR #295 (2024-02-29): ci: edit checksum artifact paths to allow automatic check
Closes #270.

All checksums (except windows) can now be verified using standard way with sha256sum utility:

1. `sha256sum <binary>`
2. `sha256sum --check <binary>-checksum`
3. `<binary>: OK`
closed 2024-02-29 by Frigyes06