Universal-Debloater-Alliance/universal-android-debloater-next-generation #283: bug(scope): Buttons missing: Import selection, export selection, Apps

Nojevah, 2024-02-22:
**Describe the bug**
Maybe it's because of the actual bug preventing updating app, but I think it will still be a problem.
"Import selection" and "export selection" are missing.
Also, when there's an update, it replaces the button "Apps"; so if you go into settings, you can't go back to the "apps" screen.

[image]

**Expected behavior**
The 3 buttons should always be visible.

Frigyes06, 2024-02-25:
This is not a bug. The UI was redesigned a while ago. Use backup selection in the review modal instead

closed 2024-02-25 by Frigyes06

Nojevah, 2024-02-25:
Sorry I don't understand your answer.
How do I export a selection in a text file, and how do I import this text file to automatically check packages from this selection ?

AnonymousWP, 2024-02-25:
You don't anymore, this is replaced by a new backup feature. I however think it's less usable than what it used to be.

Nojevah, 2024-02-25:
Sorry I still don't understand. What is this new backup feature ?
Maybe it would help describing how I use this app:
- In the past, I have uninstalled apps via adb shell or via this app (I don't use rRecommended/advanced/expert/etc. lists since they don't apply to me at all)
- then comes the first best part of this app: I export all uninstalled apps in a file
- when I reinstall a phone, I use the 2nd best part, I import the .txt file, all apps are checked, and I just uninstall all apps in 1 click.
If I can't export/import a text file, you'll understand I prefer the old uad.

Frigyes06, 2024-02-25:
> You don't anymore, this is replaced by a new backup feature. I however think it's less usable than what it used to be.

I aggree with this, but the old feature was problematic. I think we should open a feature request for "debloat lists" where it would export the names of selected packages, then try to import them when opened.

Frigyes06, 2024-02-25:
> Sorry I still don't understand. What is this new backup feature ?
> Maybe it would help describing how I use this app:
> - In the past, I have uninstalled apps via adb shell or via this app (I don't use [...]

I understand your frustration, however the GUI and underlying mechanisms had to change due to the implementation of lists. Feel free to open a feature request, describing the feature you want in as much detail as possible.

itsmartashub, 2025-10-24:
Hi,, there.

> You don't anymore, this is replaced by a new backup feature. I however think it's less usable than what it used to be.

Can't agree more.

While investigating the backup file I noticed it contains metadata such as device ID and lists of installed/enabled apps. I’d like to reuse a backup created on one device to apply changes (for example, uninstall apps) on a completely different device or ROM. A few questions about that scenario:

- Does the importer/checker enforce the original device ID or otherwise prevent applying a backup to a different device?
- How does the tool handle package mismatches? If a package is marked “enabled” (or present) in the backup but that package does not even exist on the target ROM, will the tool skip it, fail, or attempt to install/enable it?
- Are there safeguards to avoid breaking the target device (for example, by enabling or changing system apps that aren’t present on the target ROM)?
- Is there a recommended safe workflow or flags/modes (dry-run, whitelist/blacklist, apply-only-uninstall, etc.) to limit a backup application to only uninstalling apps that are present on the target device?

I don't want to risk breaking a device/ROM by blindly applying a backup from a different ROM. Any guidance or recommended steps would be appreciated. Thanks