Universal-Debloater-Alliance/universal-android-debloater-next-generation #285: bug(scope): UAD does not manage to get apps list and hide app list

Nojevah, 2024-02-23:
**Describe the bug**
When opening UAD, it doesn't list apk and their status, instead it displays a message telling us to wait.
Clicking on "No internet?" does nothing.

[image]

**Expected behavior**
Update of the list should be in background and only applied when user click on it.
Also the UAD should have an internal version in the app, for people who don't have Internet.

I know it's a temporary problem which should be fixed soon, but my bug report was to fix it for these cases of "no Internet connection".

ThatGreenBlur, 2024-02-24:
I know its just a temporary problem but for people who really really need to use uadng rn i made a build thats stuck on old version use responsibly and update soon! https://github.com/ThatGreenBlur/uad-oldlistsystem/releases/tag/v1.0.0

AnonymousWP, 2024-02-25:
> I know its just a temporary problem but for people who really really need to use uadng rn i made a build thats stuck on old version use responsibly and update soon! https://github.com/ThatGreenBlur/uad-oldlistsystem/releases/tag/v1.0.0

1.0.2 is already out: https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/releases/tag/v1.0.2

AnonymousWP, 2024-02-25:
Anyway, this issue has already been reported and is now fixed with 1.0.1. Download 1.0.2 manually.

closed 2024-02-25 by AnonymousWP

ThatGreenBlur, 2024-02-25:
> 1.0.2 is already out: https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/releases/tag/v1.0.2

i know and that was made when it was broken, ill take it down now to remove

Nojevah, 2024-02-25:
Indeed, it works now. Thanks @ThatGreenBlur for the temporary build which helped me.