Universal-Debloater-Alliance/universal-android-debloater-next-generation #299: pkg: com.sonymobile.telephony.extension

maykelbembibre, 2024-02-26:
**Your phone model:** Sony Xperia 10 V

**Packages:**

```
com.sonymobile.telephony.extension
...
```

- [ ] **I removed all those packages on my phone**
      If not why. Leave the brackets blank and explain why.

## Document each package the best you can

Hello, I bought a Sony Xperia 10 V and the battery lasts a lot, except for when the package com.sonymobile.telephony.extension starts getting a hell of a lot of wakelocks and fries my CPU. I don't know what that package does, it's not on your JSON list, either, and I would like to have some insight on whether it's safe to disable or not. It's by far the package causing the most wakelocks on my Sony phone and I don't think it is necessary, as Android itself must come with built-in telephony services or not a single phone would be able to get any reception at all!

I decompiled the APK for Android 13, which is the version I have, I have shared the Java code with you on this link from my Drive repository:
https://drive.google.com/file/d/1285KNYRACkGXyz-w-BGcSa1uLIxEoRJf/view?usp=sharing

I have tried to read it but I know nothing about telephony services or cell signal or reception. My guess is that this package should have no impact on my phone's ability to get cell reception. But, why is it installed then?

**List**: `Google`|`Misc`|`OEM` (manufacturer)|`AOSP`|`Pending`|`Carrier` (isp).

**Removal**:

- Recommended -- Pointless or outright negative packages, and/or apps available through Google Play.
- Advanced -- Breaks obscure or minor parts of functionality, or apps that aren't easily enabled/installed through Settings/Google Play. This category is also used for apps that are useful (default keyboard/gallery/launcher/music app.) but that can easily be replaced by a better alternative.
- Expert -- Breaks widespread and/or important functionality, but nothing important to the basic operation of the operating system. Removing an Expert package should not bootloop the device (unless mentioned in the description) but we can't guarantee it 100%.
- Unsafe -- Can break vital parts of the operating system. Removing an Unsafe package have an extremely high risk of bootlooping your device.

### \<package name\>

**List**: \<list\>
**Removal**: \<recommendation list\>

> Description. Link to its Playstore page if it exists.

AnonymousWP, 2024-02-26:
You didn't fill in the issue template correctly. You didn't explain why you didn't check the checkbox, and second, the list and description is missing.

maykelbembibre, 2024-02-26:
The problem is that I know nothing about that package, I only know that it can cause too many CPU wakelocks. That's why I didn't fill in the description. I have tried to find out about what it does but found nothing. I would like to know exactly what it does in order to work out whether it's safe to disable.

AnonymousWP, 2024-02-27:
Maybe @Blitz517 can help.

maykelbembibre, 2024-02-27:
I wound up taking the phone back. A package that calls itself telephony services doesn't look like safe to disable at all. And it was hogging a lot of CPU while on standby. I'm too sick of having phones full of the manufacturer's crap... I'm sad...

AnonymousWP, 2024-03-01:
@Blitz517 needs to reply whether he knows or not. He seems to be busy as he's inactive nowadays.

maykelbembibre, 2024-03-04:
Well it's been 24h since I uninstalled the package com.sonymobile.telephony.extension on my Xperia 10 V and I haven't seen any downgrade in functionality. We really should consider adding this package to the UAD list. It looks like this package contributes to cellular connectivity, but since I removed it, the stock Android phone services have managed to keep the phone correctly connected to the cellular network. I can make and receive calls and SMSs and use mobile data and the quality remains good. Also, since the removal of the package, my wakelock time has decreased A LOT. That package used to take up from 2 - 12% of the total idle time and keep the phone awake.

Blitz517, 2024-03-09:
I found some things on this app, omadm, wifi calling, IMS. its app for dual SIM reachability? Not sure if its app for debugging.

AnonymousWP, 2024-03-19:
So should we add it?

AnonymousWP, 2024-03-24:
@Blitz517 Bump.

AnonymousWP, 2024-03-29:
@maykelbembibre Does Wi-Fi calling work if disabled?