Universal-Debloater-Alliance/universal-android-debloater-next-generation #303: feat(ci): Ability to check typos in uad_lists pkg descriptions

jxdv, 2024-02-27:
**Describe the feature you want**
With the restart of UAD-ng development there's a lot of new packages being added, and also a lot of packages taken over from the old UAD. Since we're all just human and make mistakes there's a lot of typos / syntax mistakes etc..

It'd be good if we had a CI solution which would check description of each newly added package. Currently I am not aware of any such pipelines available and most likely it would have to be something regex-based.

If anyone was probably hyped enough and had a lot of free time he/she could develop a solution specifically crafted for UAD-ng, but that's probably a longshot..

Frigyes06, 2024-03-01:
Maybe something like this:
https://github.com/marketplace/actions/github-spellcheck-action

AnonymousWP, 2024-03-01:
> Maybe something like this: https://github.com/marketplace/actions/github-spellcheck-action

Doesn't support JSON it seems, see https://github.com/marketplace/actions/github-spellcheck-action#features.

nozwock, 2024-05-30:
You may look into this, [typos-action](https://github.com/marketplace/actions/typos-action)

rsb-23, 2024-11-28:
**The Crazy Idea** (but just think once)

Lets create 1 always open PR with only `uad_lists.json` - just for typos.
Then we can ask contributors/users to add a review comment for each typo instead of creating a new issue or PR.
And once a month, all typo fixes can be clubbed as single commit.
_Aim is to streamline typo issues and fixes._

---
I have not thought through the idea, but it may be automated (atleast 90%) using cherry-pick, `gh pr review`, etc.
Also, if this works - something similar can be used for package descriptions.

AnonymousWP, 2024-12-17:
I'm not a fan of that approach. Not really trackable and too hacky. Doesn't really follow best practices either, but thanks for your input nonetheless of course!