Universal-Debloater-Alliance/universal-android-debloater-next-generation #308: bug(app): Unused variable 'res' when building no-self-update

Frigyes06, 2024-03-01:
### Describe the bug

warning: unused variable: `res`
   --> src/gui/mod.rs:270:44
    |
270 |             Message::_NewReleaseDownloaded(res) => {
    |                                            ^^^ help: if this is intentional, prefix it with an underscore: `_res`
    |
    = note: `#[warn(unused_variables)]` on by default

### Expected behavior

There are no unused variables in the program

### You have a solution?

There are already parts that conditionally get excluded if building no-self-update. This code will need to be enclosed similarly too.

### Provide logs

N/a

### Acknowledgements

- [X] This issue is not a duplicate of an existing bug report.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

edgy-b, 2024-03-16:
@Frigyes06 Any other features? `cargo build --features "no-self-update"`

Probably fixed, but can you test it? Builds fine for me before fix and after fix.

edgy-b, 2024-03-16:
> @Frigyes06 Any other features? `cargo build --features "no-self-update"`
> 
> Probably fixed, but can you test it? Builds fine for me before fix and after fix.
> 
> https://github.com/edgy-b/universal-android-debloater-next-generation/tree/feat/no-self-update

nevermind, managed to reproduce, fixed

edgy-b, 2024-03-16:
Nevermind, managed to reproduce, fixed.

PR #344 (2024-03-18): fix(app): unused variable 'res' when building with no-self-update
 - fixed issues when feature: no-self-update flag was passed

resolves #308
closed 2024-03-18 by Frigyes06